apache/nuttx #2406: scanf stop parse empty strings in scanset

lukegluke, 2020-11-26:
With 350295d0098d8860ae2450bd82fdd8352bbbe98f commit in scanset conversion in lib_libvscanf.c was added:
https://github.com/apache/incubator-nuttx/blob/b021375f2c8d86020d3f67d8dceacb24b8d67ea3/libs/libc/stdio/lib_libvscanf.c#L469-L473
that stops on empty string.

Before this I can successfully parse string:
`0,"","",,"INITIAL"`
with
`sscanf(response, "%d, \"%3[^\"]\", \"%15[^\"]\", %5[^,], \"%14[^\"]\"", &index, mode, addr, port_str, state_sock)`
it returned 5. With NuttX 10.0.0 now it stops on empty string for  \"%3[^\"]\" and returns 1 (just first %d number).

At the same time there is no such condition for string conversion %s
https://github.com/apache/incubator-nuttx/blob/b021375f2c8d86020d3f67d8dceacb24b8d67ea3/libs/libc/stdio/lib_libvscanf.c#L413

Is this a bug or standard compliance?

xiaoxiang781216, 2020-11-29:
@lukegluke why not try the same code on other OS?

lukegluke, 2020-11-29:
Yeah, you're right. Sorry for bothering!

closed 2020-11-29 by lukegluke