apache/nuttx #2425: Code size growth between 9.1.0 and 10.0.0-RC0

btashton, 2020-11-27:
# Why is the Build Larger
## Intro
There was some discussion of the growth in code size between 9.1.0 and 10.0.0-RC0.  We have some tools to help us look into this, so I did a little digging.

To be clear I am not making any statement here about if we should do something about it or not, but I wanted us to at least be aware of what caused these increases.

## Discovery
For reference I am using the `stm32f4discovery:nsh` as the reference target because it seems to be a common target for people that has been around for a long time.

So first off here is the top level size information without with the standard configuration out of the box build which includes optimisation.
```
❯ arm-none-eabi-size ../testdata/opt/nuttx-*
   text	   data	    bss	    dec	    hex	filename
  69947	    104	   2104	  72155	  119db	../testdata/opt/nuttx-10.0.0-RC0
  69447	    104	   2476	  72027	  1195b	../testdata/opt/nuttx-9.1.0
```
.text sees an increase of 500 bytes and .data for 372.  These are not huge, but are also not insignificant and but we cannot keep increasing by this much every release.

There are a lot of changes between these two releases, but we can first analyse the debug information to get a better idea what is going on.  I added with kconfig-tweak the debug symbols and the  used Bloaty to generate these reports

Here is the quick script I used to generate the builds I wanted to look at:
```bash
set -e +x
NUTTX_TAG=$1
APPS=apps
NUTTX=incubator-nuttx
OUTDIR=testdata
for REPO in $NUTTX $APPS
do
  pushd $REPO
  git clean -xdf
  git checkout $NUTTX_TAG
  popd
done
pushd $NUTTX
./tools/configure.sh stm32f4discovery:nsh
kconfig-tweak --enable DEBUG_SYMBOLS
make oldconfig
make -j8
popd
mkdir -p testdata
cp $NUTTX/nuttx $OUTDIR/$NUTTX_TAG
```

This is the comparison between 10.0.0-RC0 and 9.1.0.
```
❯ bloaty -n 20 -d sections,symbols ../testdata/nuttx-10.0.0-RC0 -- ../testdata/nuttx-9.1.0
    FILE SIZE        VM SIZE    
 --------------  -------------- 
  +1.8% +16.2Ki  [ = ]       0    .debug_info
  +0.7% +1.64Ki  [ = ]       0    .debug_line
  +3.8% +1.13Ki  [ = ]       0    .debug_str
  +0.6%    +636  +0.6%    +636    .text
    +1.9%    +400  +1.9%    +400    [103 Others]
    [NEW]    +284  [NEW]    +284    mm_memalign
    [NEW]    +232  [NEW]    +232    nxtask_init
    [NEW]    +212  [NEW]    +212    nxsem_clockwait
     +83%    +158   +83%    +158    fclose
     +61%    +132   +61%    +132    nsh_session
    [NEW]     +96  [NEW]     +96    cxx_initialize
     +44%     +76   +44%     +76    nx_vioctl
     +95%     +72   +95%     +72    proc_findnode
     +19%     +64   +19%     +64    nx_vopen
    +238%     +62  +238%     +62    inode_search
    [NEW]     +62  [NEW]     +62    sem_clockwait
   -72.0%     -72 -72.0%     -72    wd_initialize
    [DEL]     -76  [DEL]     -76    up_cxxinitialize
   -46.2%     -96 -46.2%     -96    nxthread_create
   -66.7%    -120 -66.7%    -120    isspace
   -76.7%    -138 -76.7%    -138    nsh_system
[... 111 lines cut ...]
+              (uint32_t *)kumm_memalign(CONFIG_STACK_ALIGNMENT, alloc_size);
         }
 #endif /* CONFIG_TLS_ALIGNED */
 
```
This change came in as part of this PR https://github.com/apache/incubator-nuttx/pull/1562

## `proc_findnode` (+72)
This increase seems to have come in as part of this https://github.com/apache/incubator-nuttx/pull/1490
This is a minor code change to fix a bug, but the generated code is surprisingly large. 
[image]

## `fclose` (+158)
This largely comes from code complexity change that came in as https://github.com/apache/incubator-nuttx/pull/1611
Once again the change looks small, but ends up creating a fair bit of code.
[image]

## `nsh_session` (+132)
Used the same technique as before looking at the call graph.  We have a quite a bit of complexity increase here that came in with
https://github.com/apache/incubator-nuttx-apps/pull/367 which adds the ability to pass in arguments to `nsh`.  I suspect this functionality might have been best held behind a config flag.  The file stream change added some code size here, but very little.

## `nxtask_init`
(TODO)
## `nxsem_clockwait`
(TODO)

btashton, 2020-11-27:
@patacongo @xiaoxiang781216 @acassis @davids5  for awareness.  None of these seem like "bugs" and are well intentioned, but it is something we should be aware of.  The memalign one is the big one, I am not sure if there is something we can do here, I was not very involved in the TLS changes.

patacongo, 2020-11-28:
> 
> 
> @patacongo @xiaoxiang781216 @acassis @davids5 for awareness. None of these seem like "bugs" and are well intentioned, but it is something we should be aware of. The memalign one is the big one, I am not sure if there is something we can do here, I was not very involved in the TLS changes.

The build for this particular configuration increases because mm_memalign() was not included in the build before and is now drawn into the build because new code in up_create_stack() now references it.

The only way to avoid that would be to avoid calling mm_memalign() (or any of its variants in the different namespaces).

Other configurations that already used mm_memalign() would, therefore, not see any code growth but most simple configurations will certainly be impacted.  So I don't think the problem is really so significant, but does increase the base size of most minimal configurations.  That is unfortunate primarily for marketing reasons since it makes the OS seem much large in most size investigations.

patacongo, 2020-11-28:
> The only way to avoid that would be to avoid calling mm_memalign() (or any of its variants in the different namespaces).

The previous code avoided this code growth by using a simple trick:  It just allocated memory using mm_malloc() family functions and adjusted the base address and size to accomplish the alignment.  This does not add code size because mm_malloc() is always present.  mm_memalign(), on the other hand, is a larger, complex function that is not normally included in the link.

btashton, 2020-11-28:
Yeah I agree.  I mostly just wanted to put this together so there was some level of due diligence on the release and we were not just saying looks like code size is larger `¯\_(ツ)_/¯`  I run my script against some other configurations including some network related ones, and found the increase to be similar.  Most of them did not use memalign before so this is likely a cost in most configurations, and most of the file related functions that deal with streams gained a little bit of code size each which added up to ~200-300 bytes new code and was counteracted by some other clean-up in common code.

patacongo, 2020-11-28:
It would be nice to know about code size issues at the time the PR is generated vs. he 11th hour before a release.

patacongo, 2020-11-28:
Are you considering a "Code Size" section in the release note?

btashton, 2020-11-28:
> It would be nice to know about code size issues at the time the PR is generated vs. he 11th hour before a release.

I agree.  This is part of what I am working on, we now store all the binaries from the CI runs so we can do analysis on them, but I'm trying to figure out what the right thing to do to show the results.

Showing the code size change for 200 configurations in a github comment is not helpful, but I might be able to highlight configurations where there has been a high percentage change or new/deleted symbols across all the builds. 

For core OS features I imagine we can get a lot of benefit just looking at something like ostest configurations on a couple architectures.  It may not be super representative of real world configurations, but I suspect it will bring things out.

I'll see what I can come up with and then we can iterate on it.

> Are you considering a "Code Size" section in the release note?

Probably not a bad idea going forward. It also shows that it is something we are prioritizing which I think is important.

patacongo, 2020-11-28:
Hmm.. for the case of ARM with CONFIG_MM_SMALL=n, the memory returned by mm_malloc() will already be aligned correctly to 8-byte address boundaries.  That is required because the allocated memory must be capable of hold aligned long long and aligned double types in allocated structures.  So for this particular case, switching to mm_memalign() is an unfortunate waste and completely unnecessary.

davids5, 2020-11-28:
@btashton Thank you this is great engineering and analysis! 

It adds a tremendous level of understanding to the cause and effect of the changes in the release.  It would make sense to run bloaty on some representative config's in the PR. 

Here is a thought on how to present it: We could have some config permutations just for size comparison. We would enable everything (we can) and have a `bloaty.ld` linker scripts with multi Megs of memory, so we can always link. Of course these configs would need a disclaimer not to use them.

btashton, 2020-11-28:
> It adds a tremendous level of understanding to the cause and effect of the changes in the release. It would make sense to run bloaty on some representative config's in the PR.

> Here is a thought on how to present it: We could have some config permutations just for size comparison. We would enable everything (we can) and have a `bloaty.ld` linker scripts with multi Megs of memory, so we can always link. Of course these configs would need a disclaimer not to use them.

That is actually an interesting thought.  I was already looking at possibly adding some configurations that are very feature heavy to make it easier to get some more build coverage as well.  I have a configuration that I use locally on one of the STM32 with a lot of ram/flash that has net/usb/adc/sensors/display/etc... all enabled which also makes it easier for me to do functional testing without having to spend so much time flashing everything.  There are other boards like the imxrt where we could get create some functional large configurations.

I spent some time putting some more tooling together and I think I have an idea on how to go about adding some useful visualization to this so I'll share something early next week.  We actually already have over 250 runs against commits merged in the master branch each with around 590 builds collected and that is only since the end of October, so I suspect I can show some interesting graphs from that.

patacongo, 2020-11-28:
> We would enable everything (we can) ...

For cases like the mm_memalign issue, minimal configurations should code growth better than large configurations.  When new, previously unused code is drawn into a minimal configuration, the code growth can be substantial.

Are we trying to protect the minimal configurations like nsh or detect size changes in large complex configurations.

xiaoxiang781216, 2020-11-28:
Brennan,
It's a great analysis! Why not to create a tips section in document and put
this analysis there?

btashton, 2021-01-31:
I am going to close this out since there is nothing directly actionable here.  I have been capturing the artefacts for the last few months and am going to be publishing a tool to visualize them a little.  Maybe then we will be able to get that hooked into the CI more.

closed 2021-01-31 by btashton

acassis, 2021-01-31:
@btashton about `nsh_session` (+132)
Should we open an Issue to add a config flag to enable disable nsh parameter?

acassis, 2021-01-31:
> I am going to close this out since there is nothing directly actionable here. I have been capturing the artefacts for the last few months and am going to be publishing a tool to visualize them a little. Maybe then we will be able to get that hooked into the CI more.

I agree! Maybe we need to setup an automatic way to "pull the plug" when the binary increases more than 5%. Did you have some idea about how we could do it?

btashton, 2021-01-31:
> @btashton about `nsh_session` (+132)
> Should we open an Issue to add a config flag to enable disable nsh parameter?

It's more complicated then that. One symbol grew the other got smaller because of the change. This net increase is really just how memalign got pulled in which as mentioned in real configs is likely added anyway. 

Anyway we can discuss what to do with the tooling when I have it ready, right now I have just been capturing data and trying to automate some of what I did here.