apache/nuttx #2428: DEPPATH fails in the current make dependencies process(Make.dep)

anchao, 2020-11-28:
DEPPATH fails in the current make dependencies process(Make.dep),  
which makes it impossible for many submodules using DEPPATH to be incrementally compiled.
e.g: 

mm/bin:

origin(works as expected):

```
mm/bin$ cat Make.dep | grep iob_
bin/iob_add_queue.o: iob/iob_add_queue.c /usr/include/stdc-predef.h \
bin/iob_alloc.o: iob/iob_alloc.c /usr/include/stdc-predef.h \
bin/iob_alloc_qentry.o: iob/iob_alloc_qentry.c /usr/include/stdc-predef.h \
bin/iob_clone.o: iob/iob_clone.c /usr/include/stdc-predef.h \
bin/iob_concat.o: iob/iob_concat.c /usr/include/stdc-predef.h \
bin/iob_copyin.o: iob/iob_copyin.c /usr/include/stdc-predef.h \
bin/iob_copyout.o: iob/iob_copyout.c /usr/include/stdc-predef.h \
bin/iob_contig.o: iob/iob_contig.c /usr/include/stdc-predef.h \
bin/iob_free.o: iob/iob_free.c /usr/include/stdc-predef.h \
bin/iob_free_chain.o: iob/iob_free_chain.c /usr/include/stdc-predef.h \
bin/iob_free_qentry.o: iob/iob_free_qentry.c /usr/include/stdc-predef.h \
bin/iob_free_queue.o: iob/iob_free_queue.c /usr/include/stdc-predef.h \
bin/iob_initialize.o: iob/iob_initialize.c /usr/include/stdc-predef.h \
bin/iob_pack.o: iob/iob_pack.c /usr/include/stdc-predef.h \
bin/iob_peek_queue.o: iob/iob_peek_queue.c /usr/include/stdc-predef.h \
bin/iob_remove_queue.o: iob/iob_remove_queue.c /usr/include/stdc-predef.h \
bin/iob_statistics.o: iob/iob_statistics.c /usr/include/stdc-predef.h \
bin/iob_trimhead.o: iob/iob_trimhead.c /usr/include/stdc-predef.h \
bin/iob_trimhead_queue.o: iob/iob_trimhead_queue.c \
bin/iob_trimtail.o: iob/iob_trimtail.c /usr/include/stdc-predef.h \
bin/iob_navail.o: iob/iob_navail.c /usr/include/stdc-predef.h \
```

current(break):

```
mm/bin$ cat Make.dep | grep iob_
iob_add_queue.o: iob/iob_add_queue.c /usr/include/stdc-predef.h \
iob_alloc.o: iob/iob_alloc.c /usr/include/stdc-predef.h \
iob_alloc_qentry.o: iob/iob_alloc_qentry.c /usr/include/stdc-predef.h \
iob_clone.o: iob/iob_clone.c /usr/include/stdc-predef.h \
iob_concat.o: iob/iob_concat.c /usr/include/stdc-predef.h \
iob_copyin.o: iob/iob_copyin.c /usr/include/stdc-predef.h \
iob_copyout.o: iob/iob_copyout.c /usr/include/stdc-predef.h \
iob_contig.o: iob/iob_contig.c /usr/include/stdc-predef.h \
iob_free.o: iob/iob_free.c /usr/include/stdc-predef.h \
iob_free_chain.o: iob/iob_free_chain.c /usr/include/stdc-predef.h \
iob_free_qentry.o: iob/iob_free_qentry.c /usr/include/stdc-predef.h \
iob_free_queue.o: iob/iob_free_queue.c /usr/include/stdc-predef.h \
iob_initialize.o: iob/iob_initialize.c /usr/include/stdc-predef.h \
iob_pack.o: iob/iob_pack.c /usr/include/stdc-predef.h \
iob_peek_queue.o: iob/iob_peek_queue.c /usr/include/stdc-predef.h \
iob_remove_queue.o: iob/iob_remove_queue.c /usr/include/stdc-predef.h \
iob_statistics.o: iob/iob_statistics.c /usr/include/stdc-predef.h \
iob_trimhead.o: iob/iob_trimhead.c /usr/include/stdc-predef.h \
iob_trimhead_queue.o: iob/iob_trimhead_queue.c /usr/include/stdc-predef.h \
iob_trimtail.o: iob/iob_trimtail.c /usr/include/stdc-predef.h \
iob_navail.o: iob/iob_navail.c /usr/include/stdc-predef.h \
```

Regression by:

https://github.com/apache/incubator-nuttx/pull/2335
https://github.com/apache/incubator-nuttx/commit/d5b6ec450fde069a4e64569b0eb7e4fcb3b96e83

anchao, 2020-11-28:
Hi, @v01d Could you please have a look? thank you!

v01d, 2020-11-28:
hi, please see linked PR

PR #2432 (2020-11-28): build system: add missing --obj-path to MKDEP
## Summary

After last addition of parallel dependency generation, --obj-path was not passed to mkdep, which is necessary
for certain cases where the Make.dep is placed in a subdirectory.

## Impact

Fixes #2428 

## Testing

Local dependency generation of affected targets
closed 2020-11-28 by xiaoxiang781216