apache/nuttx #2443: docs: Missing instruction for building from relase budles

btashton, 2020-11-29:
The quickstart guide in the documentation does not list how to build from the Apache release bundles.
I know that as maintainers we usually work off of the git repos, we still need to make it very clear how to build via the release tarballs as well as link back to the release list on the webpage.

v01d, 2020-11-29:
I think that we can simply add the relevant commands to the second tab here: https://nuttx.apache.org/docs/latest/quickstart/install.html#get-source-code

BTW, I was thinking that the "quickstart" doesn't serve much purpose in itself and it would be better to have only the "getting started" which goes into the required level of detail to get started. Otherwise someone using just the quickstart will probably get into issues already described in the "getting started". Also, the kconfig installation is only described in "quickstart". 
I think it would be much better to just have "getting started" and have it self contained

btashton, 2020-11-29:
Yeah right now we have too many paths each which are a little different.  What made this come up was when I sent out the email to the general list for voting and realized that we still have not done much to help someone coming in from the outside especially if they don't have a lot of embedded experience.

I also am a little worried that  we added a dependency in macOS for objdump from binutils (via brew install) and not the normally included one.