apache/nuttx #2446: STM32 PWM randomly changes polarity

diegoherranz, 2020-11-30:
Following the video tutorials by @acassis (thanks for those by the way!), I've noticed that the PWM output polarity changes randomly. 

Build the stm32f103-minimum:pwm config:
```
$ make distclean
$ tools/configure.sh stm32f103-minimum:pwm
$ make
$ (flash with openocd) 
```

Call the pwm example app a few times until the PWM polarity changes. It seems random but normally not more than 10 attempts needed to make it fail. 
Inside NSH:
```
nsh> pwm -d 10
nsh> pwm -d 10 (run multiple times...)
```

After the polarity has been reversed, it can come back to non-inverted if you keep running the same command.

Enabling debug information, I've seen that the polarity bit in the CCER register does actually change.

Good:
`0pwm_dumpregs:  CCER:  0100 CNT:  d9544 PSC:   u000a ARR:   tffad`

Bad:
`0pwm_dumpregs:  CCER:  0300 CNT:  d9545 PSC:   u000a ARR:   tffad`

It seems that something is toggling that bit, but I haven't seen what. Any idea? Am I doing something wrong?

Thanks!

raiden00pl, 2020-11-30:
Could you try https://github.com/apache/incubator-nuttx/pull/2448 ?

diegoherranz, 2020-11-30:
See https://github.com/apache/incubator-nuttx/pull/2448#issuecomment-736032200

Thanks.

closed 2020-12-01 by xiaoxiang781216