apache/nuttx #2459: Out of bounds access in stm32 usb endpoint list

brymko, 2020-12-02:
In the [`stm32_usbdev_s`](https://github.com/apache/incubator-nuttx/blob/master/arch/arm/src/stm32/stm32_usbdev.c#L378) struct the length of `eplist` array is set to `STM32_NENDPOINTS`. This array length is respected throughout the initialization process. However in [`stm32_hpinterrupt`](https://github.com/apache/incubator-nuttx/blob/master/arch/arm/src/stm32/stm32_usbdev.c#L2433) an `epno` is passed to [`stm32_epdone`](https://github.com/apache/incubator-nuttx/blob/master/arch/arm/src/stm32/stm32_usbdev.c#L1607) which may not be in bounds of this array. `USB_ISTR_EPID_MASK` resolves to `0xf` which exceeds a valid array mask by a magnitude of 2. [`stm32_lptransfer`](https://github.com/apache/incubator-nuttx/blob/master/arch/arm/src/stm32/stm32_usbdev.c#L2388) displays similar behaviour.
     
If i could get confirmation about the validity of this flawed logic i'd be more than happy to submit a pull request fixing the issue at hand!

davids5, 2020-12-02:
Have you observed a error in a running system or are your comments just based on looking at the code? I ask because I think the HW is configured and will only return valid valises for the configured endpoint count.

brymko, 2020-12-02:
Hi, thanks for the quick reply! Indeed this wasn't found on a real hardware but as a combination of emulation & manual code review. However i have not found a point in where the hardware is configured that way. Thus i was looking through [this manual](https://www.st.com/resource/en/application_note/dm00296349-usb-hardware-and-pcb-guidelines-using-stm32-mcus-stmicroelectronics.pdf) (page 4-6) to stand my point. After reading your expected message i put it up once again for closer inspection and i noticed how the high endpoint boards all have their custom usb implementation where this issue is not present.  
  
If you feel like this won't be an problem in the future feel free to close this issue.

davids5, 2020-12-02:
Yes it is not "obvious" Have a look at the reference manual for a stm32F4 and then look at https://github.com/apache/incubator-nuttx/blob/master/arch/arm/src/stm32/stm32_usbdev.c#L3083 then https://github.com/apache/incubator-nuttx/blob/master/arch/arm/src/stm32/stm32_usbdev.c#L1415 and stm32_epwrite https://github.com/apache/incubator-nuttx/blob/master/arch/arm/src/stm32/stm32_usbdev.c#L1288-L1315 then https://github.com/apache/incubator-nuttx/blob/master/arch/arm/src/stm32/stm32_usbdev.c#L796

This is PMA access - think of it as dual ported memory. Only an EP that is manipulated will raise it's bit in the ISR

closed 2020-12-02 by davids5