apache/nuttx #2478: RISC-V setjmp.S is in wrong directory.

patacongo, 2020-12-04:
PR #2470 introduced RISC-V generial setjmp/longjmp for RISC-V.  Unfortunately the file  arch/risc-v/src/common/riscv_setjmp.S was place in the incorrect location.

Application code many not call directly into any code position in the kernal address space.  Violating this architectural principle does not matter very much in the FLAT build but precludes other build modes such as PROTECTED and KERNEL modes.

libs/ is the correct location for code that can be shared between user applications and OS code.  Specific, setjmp.S must be moved to libs/libc/machine/risc-v.

patacongo, 2020-12-04:
Also noted by @xiaoxiang781216 

    arch/arm/src/armv7-m/gnu/arm_setjmp.S
    arch/arm/src/armv8-m/arm_setjmp.S
    arch/sim/src/sim/up_setjmp32.S
    arch/sim/src/sim/up_setjmp64.S
    arch/sim/src/sim/up_setjmp_arm.S

btashton, 2020-12-04:
I started making this fix, and had a couple questions.

It looks like we also remove these from all the `arch/arm/<chip>/Make.defs` as this will come in via libs now?  Is there anything special that needs to happen here?
```
ifeq ($(CONFIG_ARCH_SETJMP_H),y)
ifeq ($(CONFIG_ARCH_TOOLCHAIN_GNU),y)
CMN_ASRCS += arm_setjmp.S
endif
endif
```

`arch/sim/src/sim/up_setjmp_arm.S`  -- Why is this here?  I would not expect arm specific code in the sim.

patacongo, 2020-12-04:
> It looks like we also remove these from all the 
> |arch/arm/<chip>/Make.defs| as this will come in via libs now? Is 
> there anything special that needs to happen here?
>
> |ifeq ($(CONFIG_ARCH_SETJMP_H),y) ifeq 
> ($(CONFIG_ARCH_TOOLCHAIN_GNU),y) CMN_ASRCS += arm_setjmp.S endif endif|
|For ARMv7-M, there are separate .S file for GCC and IAR.  The two 
assemblers use different syntax.|
> |arch/sim/src/sim/up_setjmp_arm.S| -- Why is this here? I would not 
> expect arm specific code in the sim. 
The newer MACs will use ARM M1 for the simulator.  Perhaps that is why 
this is there??? Or perhaps for Linux on an ARM platform?

xiaoxiang781216, 2021-04-04:
Note: the function name on sim is up_setjmp/up_longjmp, so should we rename them to setjmp/longjmp?

btashton, 2021-04-04:
@xiaoxiang781216  I think we should address the sim stuff in another pr.  The naming prefix of a lot of the sim stuff is not correct, and we should probably just track that all as one fix.

closed 2021-04-04 by btashton