apache/nuttx #2496: Amnesia:33 vulnerability in uIP-based network-stack

philip-n, 2020-12-08:
On December 8th 2020, there's been widespread reporting about a family of vulnerabilities in the network stacks of various embedded OS, called "AMNESIA:33".

The [original report](https://www.forescout.com/company/resources/amnesia33-how-tcp-ip-stacks-breed-critical-vulnerabilities-in-iot-ot-and-it-devices/) lists a number of vulnerabilities in uIP (page 10) and states that 
> The NuttX RTOS started by importing uIP, but then evolved its code independently. We could verify that at least one vulnerability still applies to NuttX.

They mention https://github.com/apache/incubator-nuttx/blob/master/include/nuttx/net/ip.h as the according file.

---

Is the vulnerability known and is there already some work going on to fix this (e.g. based on an advance-warning by the report authors)?

btashton, 2020-12-09:
@philip-n yes we worked with the authors and will be sending out an announcement shortly. In the meantime you can find nuttx-10.0.1 and nuttx-9.1.1 tags that contain the fixes.

philip-n, 2020-12-09:
Thanks for the quick answer, it's great that this has already been resolved.

Thanks to everybody for their efforts, on this particular bug as well as in general!

closed 2020-12-09 by philip-n