apache/nuttx #2499: getpid should return process id not thread id

xiaoxiang781216, 2020-12-08:


xiaoxiang781216, 2020-12-08:
@patacongo do you think we should modify getpid implementation to confirm the standard?

patacongo, 2020-12-08:
NuttX does not have separate thread IDs.  Process IDs and thread Ids are the same thing.  A pthread ID is just a pid_t that is cast to a pthread_t.

The kind of of threads used in NuttX are sometimes called "heavyweight' threads.  In the Linux 2.4 timeframe, Linux also used heavyweight threads.  pthreads were provide by the old LinuxThreads library.  https://en.wikipedia.org/wiki/LinuxThreads.  It implemented threads using the Linux clone() interface; threads were basically fork's from the parent thread with some special properties -- but each was still a process and each had its own identifying process ID.

With the old Linux threads, pthread_t was also the same as a pid_t.  So the old heavyweight, LinuxThreads were exactly the way that NuttX threads work.  When you did 'ps', threads and process should the same kind of IDs.  gettid() worked basically the same as gitpid() as they do in NuttX.

Around 2006, LinuxThreads was replaced with lightweight, Native POSIX Threads Library (NPTL).  This introduced a hierarchy and a difference in representation so that threads were implemented very differently.  'ps' now shows threads as part of hierarchy and gettid() and gitpid() return very different kinds of values.

Since NuttX follows the old LinuxThreads model, I think it is best to keep the interfaces as they are now (and as they were with LinuxThreads in Linux).

Would there be any advantage of switching to model like NPTL in NuttX?  Perhaps someday.  But for the FLAT and PROTECTED mode there would not be any advantage.  They are already very "lightweight" in the sense that they do not consume very many resources.  I don't think NPTL would improve that very much.  But for the KERNEL build with its huge overhead of mapped pages and MMU page tables, they savings might be substantial.

So there is some semantic confusion about what the process ID represents.  In

patacongo, 2020-12-08:
> group ID is overloaded. Sorry about that. NuttX does not support group security as does Linux. The group ID refers to the "task group", not to the security group.

Back to the original question.  "Should getpid() return the process ID?"  Currently it returns the thread ID of the caller.  I suppose it would, more appropriately return group ID (since the process ID of each thread should be the same, only the thread ID should be different.).

That would break a LOT of user code, I suspect.

xiaoxiang781216, 2020-12-11:
@patacongo I provide a fix here: https://github.com/apache/incubator-nuttx/pull/2518/, do you think it's resanable? If it's OK, I will review all getpid caller and change to getthid as needed.

patacongo, 2020-12-11:
> 
> 
> @patacongo I provide a fix here: #2518, do you think it's reasonable? If it's OK, I will review all getpid caller and change to getthid as needed.

It looks reasonable to me.  I made one comment:  I think that the change should still apply if CONFIG_SCHED_HAVE_PARENT is not selected.  I don't think there is any reason to implement the wrong functionality in that case.

xiaoxiang781216, 2021-03-12:
> There is some semantic confusion about what the process ID represents in NuttX. NuttX uses a "task group" to associate all of the heavyweight threads. The process ID means two things. It it the thread ID of the main thread of task group (maintained in the TCB) but it is also the task group ID of the entire collection of associated threads (maintained in the group structure, usually).
> 
> group ID is overloaded. Sorry about that. NuttX does not support group security as does Linux. The group ID refers to the "task group", not to the security group.

With https://github.com/apache/incubator-nuttx/pull/2981, group ID doesn't exist anymore, the task is repsented by the main thread proccess ID now.

closed 2022-11-17 by xiaoxiang781216