apache/nuttx #2529: version.sh does not use the "latest" tag just most recent

btashton, 2020-12-13:
There is a problem here because we currently look for the most recent tag which can be a patch release for an older series.
This results in currently 9.1.1 showing up when I would expect it to be 10.0.1

Because the tag is never on the master branch and only on the release branches I am not quite sure what the right thing to do here is.  One option would be to look for the latest release branch behind the current comment and then look for the latest tag on that.

https://github.com/apache/incubator-nuttx/blob/de1ad1fdb319c87c4b2d1405075e7cb34e120377/tools/version.sh#L89

btashton, 2020-12-13:
@xiaoxiang781216 do you have any thoughts on this?

Ouss4, 2020-12-13:
Found this: `git tag -l | sort -V | tail -1`

Ouss4, 2020-12-13:
> Found this: `git tag -l | sort -V | tail -1`
> 
> Looks like these work too: `git tag -l --sort=-v:refname | head -1` `git tag -l --sort=v:refname | tail -1`

Hmmmm... but they have the same undesirable effect for older release branches. 

Maybe a solution is to check with something like `git describe` first then go to the list of tags?

Ouss4, 2020-12-13:
> Maybe a solution is to check with something like git describe first then go to the list of tags?

Tried the following, seems to work.
```diff
diff --git a/tools/version.sh b/tools/version.sh
index fd815daa91..0a1afac697 100755
--- a/tools/version.sh
+++ b/tools/version.sh
@@ -86,7 +86,11 @@ done
 OUTFILE=$1

 if [ -z ${VERSION} ] ; then
-  VERSION=`git -C ${WD} tag --sort=taggerdate | tail -1 | cut -d'-' -f2`
+  VERSION=`git -C ${WD} describe --exact-match 2>/dev/null | tail -1 | cut -d'-' -f2`
+
+  if [ -z ${VERSION} ] ; then
+    VERSION=`git tag -l --sort=v:refname | tail -1 | cut -d'-' -f2`
+  fi

   # Earlier tags used the format "major.minor", append a "0" for a patch.
```

xiaoxiang781216, 2020-12-15:
@anchao could you take a look? the code is initially written by him.

anchao, 2020-12-16:
> Tried the following, seems to work.
>
> ```diff
> diff --git a/tools/version.sh b/tools/version.sh
> index fd815daa91..0a1afac697 100755
> --- a/tools/version.sh
> +++ b/tools/version.sh
> @@ -86,7 +86,11 @@ [...]

I recommend the approach of @Ouss4 , "v:refname" is a good choice.

btashton, 2020-12-16:
Won't that mean that if I branch off of 9.1 I would get 10.0.1

I'm wondering if we would be better off with get describe and then checking to see if the tag is greater or equal to 9.0.0 (when we started using release branches) and then fall back to latest tag.

Ouss4, 2020-12-16:
> Won't that mean that if I branch off of 9.1 I would get 10.0.1

That's guarded by the first git describe.

btashton, 2020-12-16:
> > Won't that mean that if I branch off of 9.1 I would get 10.0.1
> 
> That's guarded by the first git describe.

No, soon as you are one comment ahead it will fail.

Ouss4, 2020-12-16:
Hmmm... I see, if there is a comment ahead it will fall back to the tag sort part.

anchao, 2020-12-16:
emm... I can't give a better way to resolve this issue, 
I think the best solution is to revert the incorrect mainline tag, We need to ensure the tag continuity of the master branch,
Once the Release base branch out from the master, it means that only the RC tag can be updated.
To restore the tag version to follow the mainline, we need:

1. Revert tag nuttx-9.0.0 from:
`| * 725bdfb (tag: nuttx-9.0.0-RC1, tag: nuttx-9.0.0, nuttx/releases/9.0) tools/sethost.sh:  Correct error in setting a different host.`
TO:
`* 100bd74 (tag: nuttx-9.0.0) tools/sethost.sh:  Correct error in setting a different host.`

2. Revert tag nuttx-9.1.0 from: 
`| * e4e4cce (tag: nuttx-9.1.0-RC2, tag: nuttx-9.1.0) tools/zipme.sh: Remove the option to exclude patterns based on the VCS' "ignore" file.`
TO:
`* | 1a523e5 (tag: nuttx-9.1.0) tools/zipme.sh: Remove the option to exclude patterns based on the VCS' "ignore" file.`

3. Revert tag nuttx-10.0.0 from: 
`| | * 9f6f05d (tag: nuttx-10.0.0-RC0, tag: nuttx-10.0.0) Add release notes for NuttX-10.0.0`
TO:
`* a8b539a (tag: nuttx-10.0.0) Add release notes for NuttX-10.0.0`

Git graph:

$ git log --graph --all --decorate --oneline --simplify-by-decoration
```
...
| * e4e4cce (tag: nuttx-9.1.0-RC2, tag: nuttx-9.1.0) tools/zipme.sh: Remove the option to exclude patterns based on the VCS' "ignore" file.
...
| | * 9f6f05d (tag: nuttx-10.0.0-RC0, tag: nuttx-10.0.0) Add release notes for NuttX-10.0.0
...
| * 725bdfb (tag: nuttx-9.0.0-RC1, tag: nuttx-9.0.0, nuttx/releases/9.0) tools/sethost.sh:  Correct error in setting a different host.
```

Restore the tag:

```
git tag -d nuttx-9.0.0
git tag nuttx-9.0.0 100bd740cdb6c15d0743c3d1c940e2749a588ed5
git tag -d nuttx-9.1.0
git tag nuttx-9.1.0 1a523e5519be3d2185b809a6b1a5d84b9e01f882
git tag -d nuttx-10.0.0
git tag nuttx-10.0.0 a8b539af81d567c10b46114584fe12e1bee68293
```

Check the results:

$ git log --graph --all --decorate --oneline --simplify-by-decoration
```
...
* a8b539a (tag: nuttx-10.0.0) Add release notes for NuttX-10.0.0
...
* | 1a523e5 (tag: nuttx-9.1.0) tools/zipme.sh: Remove the option to exclude patterns based on the VCS' "ignore" file.
...
* 100bd74 (tag: nuttx-9.0.0) tools/sethost.sh:  Correct error in setting a different host.
```

If it can be restored successfully, then we can change the version acquisition to the following command:

From: 
` git -C ${WD} tag --sort=taggerdate | tail -1 | cut -d'-' -f2`
To:
`git -C ${WD} describe --tags --abbrev=0 | tail -1 | cut -d'-' -f2-3`

btashton, 2020-12-16:
The tags will never be on the master branch because of how the whole release process works they will forever be on the release branches. I think we should just go with the original proposal from @Ouss4  and we can refine it more, it at least resolves the out of order issue where people working on master are seeing an old version number.

xiaoxiang781216, 2020-12-17:
Can we hold master during the release windows except the critical bug fix? so we can make a tag directly on the mater branch after the validation(vote) pass.

btashton, 2020-12-17:
Historically that has taken a month or more and we still have release branches that have multiple tags 9.1.1 and 10.0.1 for example are tags on release branches which never would have been on master.  I think we should just start with what was originally proposed.

Ouss4, 2020-12-17:
I think holding master would kind of defies the purpose of the release branches.  One of the reasons we wanted them is to not hold contributions during the release process, which as Brennan stated can take a long time.

If you guys agree, I can create a PR with that little diff.  It fixes the immediate issue we have right now.  We'll try to come up with a better solution for the rest.

PR #2557 (2020-12-17): tools/version.sh: Fix the version retrieved from GIT.
## Summary
See #2529 for details.
## Impact
Close #2529
## Testing
Get version from master and a release branch.
closed 2020-12-17 by btashton