apache/nuttx #2567: Not possible to use SPI3 DMA1, Stream 0, Channel 0 on STM32F4

w8jcik, 2020-12-19:
In `arch/arm/src/stm32/stm32_spi.c`

This commit [cfc5b59](https://github.com/apache/incubator-nuttx/commit/cfc5b596364ddbe163009c69309ababe62309c98) from NuttX 7.27 added a check if DMA channels are defined, before enabling DMA for SPI.

```
if (priv->rxch && priv->txch)
```

Type of `rxch` is `uint8_t` and `0` is used to indicate no channel being assigned.

It is fine except if we use something like this:

```
#define DMACHAN_SPI3_RX DMAMAP_SPI3_RX_1  /* DMA1, Stream 0, Channel 0 */
#define DMACHAN_SPI3_TX DMAMAP_SPI3_TX_1  /* DMA1, Stream 5, Channel 0 */
```

`DMAMAP_SPI3_RX_1` defined in `arch/arm/src/stm32/chip/stm32f40xxx_dma.h` is `0`, so it is detected as lack of DMA channel assigned.

I found it in 7.27, didn't check it with 10, but just by looking at the code I think the issue is still in place.

davids5, 2020-12-19:
@w8jcik - Good find and thank you for the clear explanation of the issue. I will check all the drivers and would up a solution.

dagar, 2020-12-21:
Nice find, would a quick define for invalid (UINT8_MAX) suffice?