apache/nuttx #2593: Merge rv32im and rv64gc into one

xiaoxiang781216, 2020-12-23:
32bit and 64bit riscv share the same specification, it doesn't make sense to support two most identical arch by two code base.

btashton, 2020-12-23:
We should be aware that some cores do have some ISA extensions (like GAP8) I'm not sure how much that will matter.
The rv32e has different a register count and some other limitations, but I don't know of any publicly available chips using this yet.
https://github.com/lowRISC/ibex has a "micro" mode that use rv32e so that could probably used for a test on an FPGA.

xiaoxiang781216, 2020-12-23:
Yes, but these variants are minor and can be controlled by Kconfig. Since the spec is controlled by a mature committee who are considering the compatibility seriously, it expect that the potential conflict between 32bit and 64bit should be very small. On the other hand, it's better to support the experimental/predated implementation by the duplicated arch code.

btashton, 2021-01-16:
@xiaoxiang781216 I am fine with this change.  `nr5m100` was RTL was never released publically and Ken never responded, and the GAP8 while real, I think is also unused and we already heard back from the contributor that they were fine.

xiaoxiang781216, 2021-01-16:
@btashton Ok, I will found time to merge 32bit and 64bit into one.

closed 2022-01-11 by xiaoxiang781216