apache/nuttx #2594: Should we remove the support of too old and incompatible risc-v chip(gap8 and nr5m100)?

xiaoxiang781216, 2020-12-23:
To support these prehistoric chips, there are many hack code in arch/risc-v/common and arch/risc-v/rv32im.
BTW, these two chips need be compiled by specific gcc toolchain, which block them from the daily check:
https://github.com/apache/incubator-nuttx-testing/blob/master/testlist/other.dat#L25-L26
So, the quality isn't good as other riscv chip too.

hhuysqt, 2020-12-25:
Hi, I'm the first contributor of the GAP8 port, and unfortunately I'm not keeping pace with the nuttx development. I think it's okay to remove its support, because the GAP8 (pulpino) core is really out-of-date, and the architecture of its processor and peripherals are not common compared to other SMP platforms. If it hinders the project upgrades, I personally agree on removing it.

btashton, 2021-01-05:
Ken,
I know you did a lot of work getting the first RISC-V support going, and
that included the nr5m100 which I understood is based on work you did
with IQ-Analog.

I don't think the RTL ever made it public, do you think this is still of
use with all the other publicly available implementations out there today?

--Brennan

kdp1965, 2021-01-26:
We can pull out the nr5m100 support.  There are other RISC-V chips and I don't know when I would be able to get the RTL pushed to a public repo anyway.  These days I use NuttX on my Eth Zurich based ri5cy core anyway instead of the nr5m100 core.

closed 2021-03-11 by btashton