apache/nuttx #2748: u-blox nina B302

tcpipchip, 2021-01-26:
Sir, i compiled nuttx to NINA B302 (it´s a NRF52840 core) !
My questions is how to enable the USB to be the serial of shell 
thanks!

tcpipchip, 2021-01-26:
i am not getting success on CDC SERIAL (VIRTUAL USB), only with GPIO

acassis, 2021-01-26:
Hi @tcpipchip please look at nuttx/arch/arm/src/nrf52/ there is not a nrf52_usb.c so I think the support to USB is not there. Someone needs to submit it. Currently we have @v01d and @raiden00pl working on nRF52xxx families, but I don't know if they have plans to add support for USB.

If the Nordic USB IP used on nRF52840 is the same used on some other MCU, the port could be easier to do, but I don't have information about it.

v01d, 2021-01-26:
Right, USB peripheral is not yet supported, but it would be a welcome addition if you're willing to try. nRF52 arch tends to be well designed and documented.

tcpipchip, 2021-01-26:
hi @v01d 

yes, you are connect!

But gpio is enough now! Thanks!!!!

btw
https://zephryninab.blogspot.com/2021/01/programando-u-blox-nina-b302-com-nuttx.html

v01d, 2021-01-26:
Great. So I understand that you worked around this limitation, so I will close this issue. If you have further questions feel free to send them to the mailing list.

closed 2021-01-26 by v01d

tcpipchip, 2021-01-26:
where is the mailing list

v01d, 2021-01-26:
Have a look at the main README.md (https://github.com/apache/incubator-nuttx).
I just realized this info is not on the main documentation, we should fix that.

tcpipchip, 2021-01-29:
@v01d 
Matias
Wich BLE did you implemented on NRF52840 ?
I would like to do a blog about

v01d, 2021-01-29:
I'm not sure what you mean. I originally started writing a BLE link-layer from scratch for nRF52 and got slave role working quite a bit but became too much to handle. I recently added support for building nordic's controller code into NuttX (not yet merged). I'm not sure if that answers your question.

tcpipchip, 2021-01-29:
Let me see If understood, you create a ble serial ?

v01d, 2021-01-29:
No, I enabled use of Nordic BLE low-level stack.

tcpipchip, 2021-03-21:
When you have news about nrf52840, post here!

v01d, 2021-03-21:
I'm not sure what you mean. BLE is already supported via SoftDevice in case that is your question.

tcpipchip, 2021-03-22:
Sorry! I mean about sensors