apache/nuttx #2766: apps/modbus/mb_m.c: missing eMBMasterASCIIStart definition

klmchp, 2021-01-28:
Error message as shown below when enabled MB_ASCII_MASTER:
mb_m.c: In function 'eMBMasterInit':
mb_m.c:176:33: error: 'eMBMasterASCIIStart' undeclared (first use in this function); did you mean 'eMBMasterRTUStart'?
  176 |       pvMBMasterFrameStartCur = eMBMasterASCIIStart;
      |                                 ^~~~~~~~~~~~~~~~~~~
      |                                 eMBMasterRTUStart
mb_m.c:176:33: note: each undeclared identifier is reported only once for each function it appears in
mb_m.c:177:32: error: 'eMBMasterASCIIStop' undeclared (first use in this function); did you mean 'eMBMasterRTUStop'?
  177 |       pvMBMasterFrameStopCur = eMBMasterASCIIStop;
      |                                ^~~~~~~~~~~~~~~~~~
      |                                eMBMasterRTUStop
mb_m.c:178:32: error: 'eMBMasterASCIISend' undeclared (first use in this function); did you mean 'eMBMasterRTUSend'?
  178 |       peMBMasterFrameSendCur = eMBMasterASCIISend;
      |                                ^~~~~~~~~~~~~~~~~~
      |                                eMBMasterRTUSend
mb_m.c:179:35: error: 'eMBMasterASCIIReceive' undeclared (first use in this function); did you mean 'eMBMasterRTUReceive'?
  179 |       peMBMasterFrameReceiveCur = eMBMasterASCIIReceive;
      |                                   ^~~~~~~~~~~~~~~~~~~~~
      |                                   eMBMasterRTUReceive
mb_m.c:181:39: error: 'xMBMasterASCIIReceiveFSM' undeclared (first use in this function); did you mean 'xMBMasterRTUReceiveFSM'?
  181 |       pxMBMasterFrameCBByteReceived = xMBMasterASCIIReceiveFSM;
      |                                       ^~~~~~~~~~~~~~~~~~~~~~~~
      |                                       xMBMasterRTUReceiveFSM
mb_m.c:182:43: error: 'xMBMasterASCIITransmitFSM' undeclared (first use in this function); did you mean 'xMBMasterRTUTransmitFSM'?
  182 |       pxMBMasterFrameCBTransmitterEmpty = xMBMasterASCIITransmitFSM;
      |                                           ^~~~~~~~~~~~~~~~~~~~~~~~~
      |                                           xMBMasterRTUTransmitFSM
mb_m.c:183:38: error: 'xMBMasterASCIITimerT1SExpired' undeclared (first use in this function); did you mean 'xMBASCIITimerT1SExpired'?
  183 |       pxMBMasterPortCBTimerExpired = xMBMasterASCIITimerT1SExpired;
      |                                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                      xMBASCIITimerT1SExpired
mb_m.c:185:17: warning: implicit declaration of function 'eMBMasterASCIIInit'; did you mean 'eMBMasterTCPInit'? [-Wimplicit-function-declaration]
  185 |       eStatus = eMBMasterASCIIInit(ucPort, ulBaudRate, eParity);
      |                 ^~~~~~~~~~~~~~~~~~
      |                 eMBMasterTCPInit
mb_m.c: In function 'eMBMasterPoll':
mb_m.c:301:7: warning: enumeration value 'EV_MASTER_PROCESS_SUCCESS' not handled in switch [-Wswitch]
  301 |       switch (eEvent)
      |       ^~~~~~
mb_m.c:301:7: warning: enumeration value 'EV_MASTER_ERROR_RESPOND_TIMEOUT' not handled in switch [-Wswitch]
mb_m.c:301:7: warning: enumeration value 'EV_MASTER_ERROR_RECEIVE_DATA' not handled in switch [-Wswitch]
mb_m.c:301:7: warning: enumeration value 'EV_MASTER_ERROR_EXECUTE_FUNCTION' not handled in switch [-Wswitch]

xiaoxiang781216, 2021-03-28:
@klmchp it's the problem in apps repo not nuttx.

closed 2021-03-28 by xiaoxiang781216

acassis, 2021-03-28:
Hi @klmchp so long time, nice to see you around again! I just opened the issue for you: https://github.com/apache/incubator-nuttx-apps/issues/635