apache/nuttx #2823: Possible memory problems with Ethernet and CDC-ACM

michallenc, 2021-02-08:
It seems like while running [teensy-4.x:netnsh-4.1](https://github.com/apache/incubator-nuttx/blob/master/boards/arm/imxrt/teensy-4.x/configs/netnsh-4.1/defconfig) configuration (ethernet + console over CDC-ACM) a huge memory leak occurs:

```
nsh> free
             total       used       free    largest
Umem:       751616  218779424          0          0
nsh> 
```
This is state after one printed ifconfig, another one results in fault and it´s necessary to reload NuttX (sometimes the fault occurs even with the first ifconfig). Ethernet configuration with console over UART is working fine, also another configurations over CDC-ACM are ok ([lcd-4.1](https://github.com/apache/incubator-nuttx/blob/master/boards/arm/imxrt/teensy-4.x/configs/lcd-4.1/defconfig) or [can-4.1](https://github.com/apache/incubator-nuttx/blob/master/boards/arm/imxrt/teensy-4.x/configs/can-4.1/defconfig)). The problems seems to occur only after running ifconfig command in netnsh-4.1. 

Unfortunately I don´t have any other board I could connect with CDC-ACM so could someone check ethernet + console over CDC-ACM with a different board so we would know whether the problem is just Teensy related or it´s something in network? Thanks.

masayuki2009, 2021-02-09:
Although I don't have the teensy board, the latest upstream works with imrt1060-evk:netnsh.

```
NuttShell (NSH) NuttX-10.0.1
nsh> uname -a
NuttX i.MXRT1060 EVK 10.0.1 73964be24f Feb  9 2021 14:00:30 arm imxrt1060-evk
nsh> ps
  PID PRI POLICY   TYPE    NPX STATE    EVENT     SIGMASK   STACK COMMAND
    0   0 FIFO     Kthread N-- Ready              00000000 002048 Idle Task
    1 100 RR       Kthread --- Waiting  Signal    00000000 002036 lpwork
    2 100 RR       Task    --- Running            00000000 002036 init
    3 100 RR       Task    --- Waiting  Semaphore 00000000 002012 Telnet daemon 0x2020e800
    4 100 RR       Task    --- Waiting  Semaphore 00000000 002012 Telnet daemon 0x2020f4c0
nsh> free
             total       used       free    largest
Umem:       735136      16048     719088     719088
nsh> ifconfig
eth0	Link encap:Ethernet HWaddr 00:e0:de:ad:be:ef at UP
	inet addr:10.0.0.2 DRaddr:10.0.0.1 Mask:255.255.255.0
	inet6 addr: fc00::2/112
	inet6 DRaddr: fc00::1/112

	RX: Received Fragment Errors  
	    00000006 00000000 00000000
	    IPv4     IPv6     ARP      Dropped 
	    00000004 00000000 00000002 00000000
	TX: Queued   Sent     Errors   Timeouts
	    00000000 00000000 00000000 00000000
	Total Errors: 00000000

             IPv4  IPv6   TCP   UDP  ICMP  ICMPv6
Received     0004  0000  0000  0004  0000  0000
Dropped      0000  0000  0000  0000  0000  0000
  IPv4        VHL: 0000   Frg: 0000
  IPv6        VHL: 0000
  Checksum   0000  ----  0000  0000  ----  ----
  TCP         ACK: 0000   SYN: 0000
              RST: 0000  0000
  Type       0000  0000  ----  ----  0000  0000
Sent         0000  0000  0000  0000  0000  0000
  Rexmit     ----  ----  0000  ----  ----  ----
nsh> renew eth0
renew [5:100]
nsh> ifconfig
eth0	Link encap:Ethernet HWaddr 00:e0:de:ad:be:ef at UP
	inet addr:192.168.10.12 DRaddr:192.168.10.1 Mask:255.255.255.0
	inet6 addr: fc00::2/112
	inet6 DRaddr: fc00::1/112

	RX: Received Fragment Errors  
	    00000009 00000000 00000000
	    IPv4     IPv6     ARP      Dropped 
	    00000006 00000000 00000003 00000000
	TX: Queued   Sent     Errors   Timeouts
	    00000002 00000002 00000000 00000000
	Total Errors: 00000000

             IPv4  IPv6   TCP   UDP  ICMP  ICMPv6
Received     0006  0000  0000  0006  0000  0000
Dropped      0000  0000  0000  0000  0000  0000
  IPv4        VHL: 0000   Frg: 0000
  IPv6        VHL: 0000
  Checksum   0000  ----  0000  0000  ----  ----
  TCP         ACK: 0000   SYN: 0000
              RST: 0000  0000
  Type       0000  0000  ----  ----  0000  0000
Sent         0002  0000  0000  0002  0000  0000
  Rexmit     ----  ----  0000  ----  ----  ----
nsh> free
             total       used       free    largest
Umem:       735136      16048     719088     719088
nsh> 

```

michallenc, 2021-02-09:
Hi @masayuki2009, I think imrt1060-evk:netnsh doesn´t use CDC-ACM settings, it communicates via serial port (there is a converter on board as I recall, so from NuttX side it´s just a serial console). Teensy with serial console over UART also works fine, it´s basically the same configuration as imxrt1060-evk. The problem only occurs if I have console over CDC-ACM.

masayuki2009, 2021-02-10:
> Hi @masayuki2009, I think imrt1060-evk:netnsh doesn´t use CDC-ACM settings, it communicates via serial port (there is a converter on board as I recall, so from NuttX side it´s just a serial [...]

Hi @michallenc,

Perhaps, you are right.
Actually, I can reproduce the problem with imxrt1060-evk:netnsh by enabling the CDC-ACM and USB CONSOLE.
I'm still not sure why the problem happens but I noticed that the problem does not happen if I remove ```CONFIG_NET_STATISTICS=y``` from defconfig.

michallenc, 2021-02-12:
@masayuki2009 Sorry for the late answer, my main focus during past few days were [changes to flexcan driver](https://github.com/michallenc/incubator-nuttx/tree/can-rewrite).
Removing `CONFIG_NET_STATISTICS=y` does indeed adrress the problem. Interestingly, `CONFIG_NET_STATISTICS=y` doesn´t seem to cause any problem for CAN. I´ll take a closer look at what´s happenign when net statistics are on.

masayuki2009, 2021-02-12:
Hi @michallenc 

I suspect imxrt_usbdev.c might have issues with D-cache.
Actually, if I add ```+CONFIG_ARMV7M_DCACHE=n ``` to the defconfig, the issue does not happen.

patacongo, 2021-02-12:
The general rules are here: 
https://cwiki.apache.org/confluence/display/NUTTX/Porting+Drivers+to+the+STM32+F7

michallenc, 2021-02-12:
@masayuki2009 Good spot! This might be actually related to [chip errata ERR011572](https://www.zlgmcu.com/data/upload/file/Utilitymcu/IMXRT1060-kanwu.pdf). Problems can occur while addresing data cache with write through. I tried to add `+CONFIG_ARMV7M_DCACHE_WRITETHROUGH=n` as switching to write back is recomended in errata, but imxrt_enet.c doesn´t support it yet.