apache/nuttx #2862: samv71-xult board support for GPIO drivers and the GPIO example application.

HBL-001, 2021-02-17:
Hi,
I am starting a new project that uses the ATMEL samv71-xult board under Nuttx.  To test my hardware I need to access the PIO lines on the EXT1 and EXT2 expansion ports, this sounds easy.  I went into menuconfig and selected DEVICE DRIVERS, then GPIO driver and then over to APPLICATION-->EXAMPLES where I selected the "GPIO Driver Example".  I save this off and flashed my board.

The GPIO Application is on the menu when I ran "help", but when I looked in /DEV there were no PIO devices there.

Has this board got support for configuring and accessing the GPIO pins through the NSH?

Kind regards

Richard

btashton, 2021-02-17:
You need to make sure your board is creating a GPIO driver like this.
https://github.com/apache/incubator-nuttx/blob/master/boards/risc-v/bl602/bl602evb/src/bl602_gpio.c#L544

patacongo, 2021-02-17:
You need to add logic to your board src/ directory to create the lower half gpio driver and to register the specific pins that you want to control.  See references to CONFIG_DEV_GPIO and CONFIG_GPIO_LOWERHALF under boards/.  See also apps/examples/gpio and include/nuttx/ioexpander/gpio.h.

HBL-001, 2021-02-18:
> You need to make sure your board is creating a GPIO driver like this.
> https://github.com/apache/incubator-nuttx/blob/master/boards/risc-v/bl602/bl602evb/src/bl602_gpio.c#L544

Hi, 
thanks for your help. 

No I had a good look aroun the    /boards/arm/samv7/samv71-xult/src/ directory and there is no driver with a name like sam_gpio.c, so it looks like I need to try to put one together.  Funnily the "autoleds" which accesses two GPIO's that are routed via LEDS refers to a header called "sam_gpio.h", which doesnt actually exist.   

Kind regards

Richard

HBL-001, 2021-02-18:
> You need to add logic to your board src/ directory to create the lower half gpio driver and to register the specific pins that you want to control. See references to CONFIG_DEV_GPIO and CONFIG_GPIO_LOWERHALF under boards/. See also apps/examples/gpio and include/nuttx/ioexpander/gpio.h.

Thank you for this help.  I have started to work out how this all goes together and life is much better than trying to add a customisation in Linux, a lot fewer places and files to gather together.

So I need to create a Sam_gpio.c file that calls Sam_configgpio()  for each of the pins I need to use.  Is this the "lower half" driver?

One set of useful pins is defined in Sam_V71_xult.h and that is a good place to start.

patacongo, 2021-02-18:
> So I need to create a Sam_gpio.c file that calls Sam_configgpio() for each of the pins I need to use. Is this the "lower half" driver?

That is only part of what you have to do.  You need to create a lower half driver that implements the interface defined include/nuttx/ioexpander/gpio.h and then you need to call gpio_pin_register() to register the GPIO driver.  It looks like there may be examples here:

    boards/arm/imxrt/imxrt1020-evk/src/imxrt_gpio.c
    boards/arm/imxrt/imxrt1050-evk/src/imxrt_gpio.c
    boards/arm/imxrt/imxrt1060-evk/src/imxrt_gpio.c
    boards/arm/samd5e5/metro-m4/src/sam_gpio.c
    boards/arm/stm32/nucleo-f429zi/src/stm32_gpio.c
    boards/arm/stm32/stm32f103-minimum/src/stm32_gpio.c
    boards/arm/stm32f0l0g0/nucleo-g070rb/src/stm32_gpio.c
    boards/arm/stm32f7/nucleo-144/src/stm32_gpio.c
    boards/arm/stm32h7/nucleo-h743zi/src/stm32_gpio.c
    boards/arm/stm32l4/nucleo-l432kc/src/stm32_gpio.c
    boards/arm/stm32l4/nucleo-l476rg/src/stm32_gpio.c
    boards/risc-v/bl602/bl602evb/src/bl602_gpio.c
    boards/sim/sim/sim/src/sim_gpio.c
    boards/xtensa/esp32/esp32-wrover-kit/src/esp32_gpio.c

The example you are looking at just configures some pins, it does not set up the gpio driver.

btashton, 2021-02-18:
Yeah the one I linked is fairly recently implemented and tested.  Most of the code in there is for supporting interrupts which you may or may not want to use. The example does demonstrate it.

HecticSalad, 2021-03-23:
I'm having some trouble doing this for the nucleo-144 board type. I'm using the stm32f767zi nucleo board, and when I try to run the example, the drivers don't show up under /dev/ .. any idea what might be going wrong? I'm enabling the GPIO driver and the GPIO example in the menuconfig

HBL-001, 2021-03-24:
> Links:

Hi,

Unfortunately you will need to write a driver to be able to use the 
GPIO.  See earlier posts.

Cheers

Richard

closed 2021-03-28 by xiaoxiang781216

HecticSalad, 2021-03-29:
For those that might be having the same issue with the nuclei-144 boards, I managed to get it going by selecting the following from menu config:

-> Application Configuration
   -> NSH library
      -> Have architecture-specific initialization 

As it turns out, the board_app_initialise() method is only called when the nsh_initialise() method calls boardctl() which requires a macro to be set first.