apache/nuttx #2909: Convert TODO list to issues

Ouss4, 2021-02-24:
Extracted from #2899

> Perhaps a better thing to do would be to convert all items in the TODO list to either issues or to Wiki pages???

I think it's better to have those TODOs as issues here in Github.  Or maybe just one pinned issue with multiple items.

@btashton @v01d any tool up your sleeves that can automate that?

Ouss4, 2021-02-24:
> Or maybe just one pinned issue with multiple items.

That list is too long to be in one issue.  So, maybe we can have the first introductory paragraph as a pinned issue with links to the other detailed descriptions in other issues.

v01d, 2021-02-24:
> Extracted from #2899
>
>
> I think it's better to have those TODOs as issues here in Github. Or maybe just one pinned issue with multiple items.
>
> @btashton @v01d any tool up your sleeves that can [...]

I think that GH's CLI tipo can let you script that.

v01d, 2021-03-30:
Do you want me to have an attempt at this? I think I should be able to make a script that parses the TODO list and runs `gh` to open one issue per entry. I could add some kind of identifier in the title and/or special label so that we distinguish them from the other ones. 
However, note that there are currently 110 issues on that file, so it would double our current issue count.

patacongo, 2021-03-30:
> Do you want me to have an attempt at this? I think I should be able to make a script that parses the TODO list and runs `gh` to open one issue per entry. I could add some kind of identifier in the title and/or special label so that we distinguish them from the other ones.

Some of the items have already been converted to issues.  Some are already "closed" and are retained in the TODO list only for information (such as the fact that memory is not freed when a task exits).  Some are probably just to trivial and old and not useful (like some of the errors reported in apps/examples).

I think you could sorts the issues by my user name to find the ones that have already been converted.  removal of the trivial things that are not worth carrying would not be too difficult.

v01d, 2021-03-30:
I think just creating all and closing the ones already addressed will probably easier than making exceptions in the issue creating script. I can removed the closed ones, but sifting for ones already addressed here is much more complex than just closing them.

If you confirm that you are OK with me creating a storm of ~100 issues I can move forward.

btashton, 2021-04-05:
@v01d if you do that can you put a label on them so that we can triage them a little to see if they may have already been resolved?

v01d, 2021-04-05:
Ok, conversion is done. I've added a special "Migrated TODO" label to easily filter these. I'll start closing those marked as "closed".

Ouss4, 2021-04-05:
Oh! thanks for doing that @v01d!

v01d, 2021-04-05:
No problem. Felt bad about all the notification emails it must have triggered though.

Ouss4, 2021-04-05:
> No problem. Felt bad about all the notification emails it must have triggered though.

Not for me. :) I've got Gmail filtering them out.

This issue should be closed then.

closed 2021-04-05 by Ouss4

fjpanag, 2021-04-06:
Some issues refer to the apps instead of the kernel.  
Maybe a manual quick check is needed now, to move them at https://github.com/apache/incubator-nuttx-apps ?

v01d, 2021-04-06:
I did move some of them. Which ones do you refer?

fjpanag, 2021-04-06:
Oh sorry, I didn't take note of them, but I think some were referring to NSH functionality etc.

If you did actually check, don't be bothered again. I will do it. And report here if I find anything misplaced.

v01d, 2021-04-06:
Sure, no problem. I may have missed a few. You can comment on the relevant issue and I can move it if needed.

PR #3439 (2021-04-05): Removing TODO file as issues are now tracked in GitHub
## Summary

As per title. Closes #2909 

## Impact

None

## Testing

N/A