apache/nuttx #2988: STM32L151 I2C problems

ReimuNotMoe, 2021-03-06:
I'm currently porting NuttX to a new (not publicly released yet) board. It uses STM32L151C8T6-A MCU.

However, I can't properly read data from I2C devices using the i2ctool.

I can list attached devices using `i2c dev 00 7f`, but dumping registers using `i2c dump -b 1 -a <addr> 256` will return garbage data. Sometimes it returns all `ff` and sometimes it returns random values.

I tried messing with the menuconfig options: Alternate implementation, dynamic timeouts, huge timeout values, etc. And none of these worked.

These I2C devices work well with STM32 HAL APIs and a Raspberry Pi.

Please tell me how to solve this problem, thanks in advance.

acassis, 2021-03-06:
Hi @ReimuNotMoe what device are you trying to dump? Do you have a Logic Analyzer to see the bus? Even a "cheaper" logic analyzer can do the trick: https://acassis.wordpress.com/2015/07/19/sigrok-and-a-cheap-saleae-clone-can-save-your-life-i-meant-your-time/

ReimuNotMoe, 2021-03-14:
> Hi @ReimuNotMoe what device are you trying to dump? Do you have a Logic Analyzer to see the bus? Even a "cheaper" logic analyzer can do the trick: https://acassis.wordpress.com/2015/07/19/sigrok-and-a-cheap-saleae-clone-can-save-your-life-i-meant-your-time/
> This way you can compare the i2ctool command from NuttX against STM32 HAL and the Raspberry Pi i2ctool.

I bought a DSLogic Plus logic analyzer and used it to inspect the i2c lines.

It seems that there are only a bunch of STARTs and STOPs and there are no actual data transfers when running `i2c dump -b 1 -a <addr> 256`.
[image]

[image]

ReimuNotMoe, 2021-03-14:
So after looking these screenshots again and again, I found the problem: the NuttX i2ctool isn't sending register addresses to the device? Is this intentional or standardized?

acassis, 2021-03-14:
Hi @ReimuNotMoe I just ordered some MCP23017 modules to test. It is not clear what is the expected behavior of the i2ctool dump.

ReimuNotMoe, 2021-03-15:
> Hi @ReimuNotMoe I just ordered some MCP23017 modules to test. It is not clear for me what is the expected behavior of the i2ctool dump.

I mean the NuttX i2ctool isn't telling the device which register to read from. It didn't write register address to device before initializing incremental read.

Since NuttX is trying to be Linux-ish in all ways (AFAIK), it might be better to make the NuttX i2ctool behave like the one in Linux.

acassis, 2021-03-15:
Yes, you are right @ReimuNotMoe the NuttX i2ctool needs to duplicate the Linux i2ctool. People can see NuttX as a Linux-like for microcontroler! ;-)

xiaoxiang781216, 2021-03-28:
Let's close this issue now, @ReimuNotMoe you can provide a patch to make NuttX i2c-tools more compatible with Linux's version.

closed 2021-03-28 by xiaoxiang781216