apache/nuttx #3109: MIMXRT106x FlexSPI driver missing

jturnsek, 2021-03-19:
Is there an implementation of MIMXRT10xx FlexSPI driver for reading and writing external flashes. I can see the code for setting flash to XIP, but the driver itself is missing. I would like to mount a filesystem on external QSPI flash connected to FlexSPI peripheral on the MIMXRT1062-EVK board.

xiaoxiang781216, 2021-03-28:
You may send the question to dev@nuttx.apache.org since many developer don't read github issue daily.

closed 2021-03-28 by xiaoxiang781216

acassis, 2021-03-28:
@jturnsek I'm got to FlexSPI driver from @ivanucherdzhiev and I'm clearing it to submit to mailing. Sorry I didn't see this issue before. I agree with @xiaoxiang781216 it is better to submit this question to dev@ mailing list anyway.

jturnsek, 2021-03-29:
Hi, thanks for the reply. Where can I get hold of a driver. I have already started to implement my own, but I would not like to discover hot water.

acassis, 2021-03-29:
> Hi, thanks for the reply. Where can I get hold of a driver. I have already started to implement my own, but I would not like to discover hot water.

I just sent it to your email. Could you help me clean it up to submit it to the mainline?

ivanucherdzhiev, 2021-03-29:
Hi,

I am sorry that i didnt had the time to clean it up and upload it my self,
but let me know if i can help with that.

best regards,
Ivan