apache/nuttx #3140: Nucleo STM32F767ZI GPIO Example

HecticSalad, 2021-03-22:
Hey there,

I'm having some trouble with the GPIO driver example on the STM32F767ZI discovery board. I feel like this should be simple but it's not working.

What I'm doing:

Cloned Nuttx including tools, apps etc.
Ran ./tools/configure.sh nucleo-144:f767-nsh
Ran make menuconfig
Ensured that Device Drivers -> IO Expander/GPIO Support -> GPIO Driver was selected (max num signals = 3)
Ensured that Application Configuratio -> Examples -> GPIO Driver Example is selected
Flashed nuttx.bin to my device using sudo openocd -f interface/stlink-v2.cfg -f target/stm32f7x.cfg -c init -c "reset halt" -c "flash write_image erase nuttx.bin 0x08000000"
Opened minicom and reset the board so that the NSH displays.
Run ls /dev to view the devices and all I see is console, null and ttyS0
Run ? and I can see the gpio example app and can run it.
The Issue:

According to: https://www.youtube.com/watch?v=VXsTLzI6idA, I should be seeing /dev/gpin0, /dev/gpint2 and /dev/gpout1, or something similar, but I'm not seeing that and so I can't use the gpio example.

Is there something I'm doing wrong? Any help would be greatly appreciated!

xiaoxiang781216, 2021-03-28:
Could you set a debug break here:
https://github.com/apache/incubator-nuttx/blob/master/boards/arm/stm32f7/nucleo-144/src/stm32_gpio.c#L285-L346
to ensure the initialization code get called.

btashton, 2021-04-05:
@HecticSalad To use the GPIO demo you need to enable these flags:
```
CONFIG_BOARD_LATE_INITIALIZE=y
CONFIG_DEV_GPIO=y
CONFIG_EXAMPLES_GPIO=y
CONFIG_NSH_ARCHINIT=y
```

You will then have these 9 pins mapped to the driver
https://github.com/apache/incubator-nuttx/blob/master/boards/arm/stm32f7/nucleo-144/src/nucleo-144.h#L183-L203

Here is it running:
```
NuttShell (NSH) NuttX-10.0.1
nsh> ls /dev
/dev:
 console
 gpin0
 gpin1
 gpin2
 gpin3
 gpint8
 gpout4
 gpout5
 gpout6
 gpout7
 null
 ttyS0
nsh> gpio -o 1 /dev/gpout4
Driver: /dev/gpout4
  Output pin:    Value=1
  Writing:       Value=1
  Verify:        Value=1
nsh> 
```

HecticSalad, 2021-04-05:
Sorry for getting back so late, work has been hectic. That basically got it working! I needed to set CONFIG_NSH_ARCHINIT because it wasn’t running the initialization. 

Thanks for the help!

closed 2021-04-05 by HecticSalad