apache/nuttx #3142: Board configs via base defconfig + supplemental options

v01d, 2021-03-22:
This is a proposal for reworking how we handle board configs in order to simplify maintenance by greatly reducing the number of configurations to keep in sync and at the same time facilitate enabling/disabling "features" (consiting of a series of config options to be set to some value) which can be more independent of a given board/arch.

The idea is to have one main defconfig (although having more than one would still be possible, if really required) for a board, which sets the minimal set of options for that board (e.g. configs related to plaform) and nothing else. Then, the build system would support applying a set of options from a separate file (e.g. `nsh.conf`) with the minimal set of features required to enable NSH (in some general purpose configuration, for example). This `.conf` file is intended to be hand-editable. 

This means that such `.conf` files can be stored somewhere only once and not repeated for every board since it does not depend on it (there could be `.conf` related to example apps, or `.conf` related to common features such as debugging). Other .conf files may actualy depend on hardware and can thus be stored inside the board directory. As a result, the number of configs in each board would be greatly reduced, which minimizes maintenance issues. Also, updating board-independent .conf files does not require "refreshing" all configs.

The procedure to include this in the build would possibly be done by extending the configure.sh script so that it does the defconfig as usual, and then applies the `.conf` file by calling kconfig-tweak over each line of this file (and then `make oldconfig` or similar to finish the configuration).

Ouss4, 2021-05-04:
> The procedure to include this in the build would possibly be done by extending the configure.sh script so that it does the defconfig as usual, and then applies the .conf file by calling kconfig-tweak over each line of this file (and then make oldconfig or similar to finish the configuration).

There is the `kconfig-merge` tool that combines multiple defconfigs into one (.config).  This procedure can be reduced to calling `kconfig-merge` for all the needed files.  Something like
```
kconfig-merge -m nsh.conf stm32.conf net.conf
make olddefconfig
```

The `make olddefconfig` and the -m option of `kconfig-merge` can be omitted if we implement `alldefconfig`.

```patch
diff --git a/tools/Makefile.unix b/tools/Makefile.unix
index 549d1be484..42328ca363 100644
--- a/tools/Makefile.unix
+++ b/tools/Makefile.unix
@@ -490,6 +490,9 @@ oldconfig: apps_preconfig
 olddefconfig: apps_preconfig
        $(Q) APPSDIR=${CONFIG_APPS_DIR} EXTERNALDIR=$(EXTERNALDIR) kconfig-conf --olddefconfig Kconfig

+alldefconfig: apps_preconfig
+       $(Q) APPSDIR=${CONFIG_APPS_DIR} EXTERNALDIR=$(EXTERNALDIR) kconfig-conf --alldefconfig Kconfig
+
 menuconfig: apps_preconfig
        $(Q) APPSDIR=${CONFIG_APPS_DIR} EXTERNALDIR=$(EXTERNALDIR) kconfig-mconf Kconfig

diff --git a/tools/Makefile.win b/tools/Makefile.win
index 9ab3427eee..0e38df2792 100644
--- a/tools/Makefile.win
+++ b/tools/Makefile.win
@@ -445,6 +445,9 @@ oldconfig: apps_preconfig
 olddefconfig: apps_preconfig
        $(Q) set APPSDIR=$(patsubst "%",%,${CONFIG_APPS_DIR})& set EXTERNALDIR=$(EXTERNALDIR)& kconfig-conf --olddefconfig Kconfig

+alldefconfig: apps_preconfig
+       $(Q) set APPSDIR=$(patsubst "%",%,${CONFIG_APPS_DIR})& set EXTERNALDIR=$(EXTERNALDIR)& kconfig-conf --alldefconfig Kconfig
+
 menuconfig: configenv apps_preconfig
        $(Q) set APPSDIR=$(patsubst "%",%,${CONFIG_APPS_DIR})& set EXTERNALDIR=$(EXTERNALDIR)& kconfig-mconf Kconfig
```

v01d, 2021-05-05:
Thanks for the info @Ouss4

btashton, 2021-05-05:
One thing that is not clear to me is how you would ever be able to refresh these configs. That seems like it would become a very manual job and in some cases cod be impossible as a option could end up being automatically selected depending on what it is merged with.

v01d, 2021-05-05:
I think that would be related to how to easily create these .conf files. While the idea is that you should be able to manually create one, it should also be possible to create one via menuconfig. For example, you could generate the .conf file from the diff between the current .config and the one being saved with options changed. This could then be automated to refresh each .conf file by loading the "base defconfig" and doing something like olddefconfig and take the diff to the base one to regenerate the .conf file.

In any case, I think that having a few .conf files needing a potential manual fix would be much more manageable than having to keep all board's defconfigs fresh. For example, some default in nsh changes, you would probably have to fix only the nsh related .conf files.

BTW, I wouldn't propose to do this with current build system. Using CMake it would be much easier, so I'd leave this be for sometime in the future.