apache/nuttx #3209: Unable to build for NUCLEO-L432KC under Arch Linux using ARM GNU toolchain in package repository

NeoChen1024, 2021-03-28:
Distro: Arch Linux
GCC -v output:
```
Using built-in specs.
COLLECT_GCC=arm-none-eabi-gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/arm-none-eabi/10.2.0/lto-wrapper
Target: arm-none-eabi
Configured with: /build/arm-none-eabi-gcc/src/gcc-10.2.0/configure --target=arm-none-eabi --prefix=/usr --with-sysroot=/usr/arm-none-eabi --with-native-system-header-dir=/include --libexecdir=/usr/lib --enable-languages=c,c++ --enable-plugins --disable-decimal-float --disable-libffi --disable-libgomp --disable-libmudflap --disable-libquadmath --disable-libssp --disable-libstdcxx-pch --disable-nls --disable-shared --disable-threads --disable-tls --with-gnu-as --with-gnu-ld --with-system-zlib --with-newlib --with-headers=/usr/arm-none-eabi/include --with-python-dir=share/gcc-arm-none-eabi --with-gmp --with-mpfr --with-mpc --with-isl --with-libelf --enable-gnu-indirect-function --with-host-libstdcxx='-static-libgcc -Wl,-Bstatic,-lstdc++,-Bdynamic -lm' --with-pkgversion='Arch Repository' --with-bugurl=https://bugs.archlinux.org/ --with-multilib-list=rmprofile
Thread model: single
Supported LTO compression algorithms: zlib zstd
gcc version 10.2.0 (Arch Repository)
```

ld -v output:
```
GNU ld (GNU Binutils) 2.36.1
```

Log:
```
echo "LD: nuttx"
LD: nuttx
arm-none-eabi-ld --entry=__start -nostartfiles -nodefaultlibs -g -T/usr/neo_chen/vcs/NuttX/nuttx/boards/arm/stm32l4/nucleo-l432kc/scripts/l432kc.ld  -L"/usr/neo_chen/vcs/NuttX/nuttx/staging" -L"/usr/neo_chen/vcs/NuttX/nuttx/arch/arm/src/board" -L "/usr/lib/gcc/arm-none-eabi/10.2.0/thumb/v7e-m+fp/hard" -L "/usr/lib/gcc/arm-none-eabi/10.2.0/../../../../arm-none-eabi/lib/thumb/v7e-m+fp/hard" \
        -o "/usr/neo_chen/vcs/NuttX/nuttx/nuttx"   \
        --start-group -lsched -ldrivers -lboards -lc -lmm -larch -lxx -lapps -lcrypto -lfs -lbinfmt -lboard -lgcc -lm --end-group
arm-none-eabi-ld: Error: unable to disambiguate: -nostartfiles (did you mean --nostartfiles ?)
make[1]: *** [Makefile:156: nuttx] Error 1
make[1]: Leaving directory '/usr/neo_chen/vcs/NuttX/nuttx/arch/arm/src'
make: *** [tools/Makefile.unix:422: nuttx] Error 2
```

Ouss4, 2021-03-28:
I've seen this issue as well on Manjaro.. I ended up downloading from [1]  instead.

1. https://developer.arm.com/-/media/Files/downloads/gnu-rm/10-2020q4/gcc-arm-none-eabi-10-2020-q4-major-x86_64-linux.tar.bz2?revision=ca0cbf9c-9de2-491c-ac48-898b5bbc0443&la=en&hash=68760A8AE66026BCF99F05AC017A6A50C6FD832A

NeoChen1024, 2021-03-28:
But this happens to AVR Toolchain too, it seems newer version of GNU ld dropped these parameters

Ouss4, 2021-03-28:
Could be the case, I had other similar LD issues with other options (in building something not related to NuttX).  Do you have any release notes that explain these new modification?

btashton, 2021-03-29:
The latest release still seems to have them and they are quite standard linker flags, I suspect something else is going on here
https://gcc.gnu.org/onlinedocs/gcc-10.2.0/gcc/Link-Options.html#Link-Options

NeoChen1024, 2021-03-29:
Maybe something's wrong with Arch Linux's cross toolchain

closed 2021-04-03 by xiaoxiang781216

hartmannathan, 2021-06-10:
This is caused by Binutils being updated to the 2.36.x line.

This is fixed by PR #3836 

This is now documented at Release Notes for the (future) NuttX-10.2 release:

[https://cwiki.apache.org/confluence/display/NUTTX/NuttX+10.2#ld-now-called-through-gcc](https://cwiki.apache.org/confluence/display/NUTTX/NuttX+10.2#ld-now-called-through-gcc)