apache/nuttx #3241: Bad stack size on task creation

patacongo, 2021-03-30:
It appears that the stack size used in up_create_stack and up_use_stack includes the size of the TLS data.  This seems to be the case in all architectures except for ARM where the issue has been fixe.  For ARM,
```
  tls_size   = INT32_ALIGN_UP(sizeof(struct tls_info_s));
  alloc_size = STACK_ALIGN_UP(stack_size + tls_size);
  stack_size = alloc_size - tls_size;
```
Where alloc_size is the size of the allocated memory and stack_size is the raw, unaligned size that will be saved and which will be colorized.

For other architectures:
```
  stack_size += sizeof(struct tls_info_s);
```
Where stack_ size is the size of the allocated memory and is also the raw, unaligned size that will be saved and which will be used for stack checking.  Am I missing something here?

patacongo, 2021-03-30:
Hmmm... this is probably not an issue for some architectures:  up_check_stack)_ and friends are only implemented only for ARM, AVR, OR1K, RISC-V, sim, xtensa.  For the others, the saved, valid stack size is wrong, but it really doesn't matter because it is not used for stack checking.

RISC-V seems to implement a fix differently from that of ARM but AVR, OR1K, sim, and Xtensa could be an issue.

v01d, 2021-03-30:
I think this would be a duplicate of #1631
If so, we can close one of the two.

closed 2021-04-07 by xiaoxiang781216