apache/nuttx #3276: tcpecho example waiting for ACK after sending [SYN, ACK] & Ping command misinterpreting sender IP

HecticSalad, 2021-04-02:
Hey there, I have two issues that I wanted to put up here:

**_Issue 1 - Ping command misinterpreting sender IP_**

Hardware: STM32F767ZI nucleo development board
What I'm doing:  
1. Cloned fresh nuttx and apps
2. Ran `./tools/configure.sh nucleo-144:f767-netnsh`
3. Used `make menuconfig` to enable network debugging
4. Ran `make`
5. Flashed to board using openocd
6. Run minicom and reset board
7. Check default IP address (10.0.0.2 is the static of the nucleo-144, so I connected the board directly to my Ubuntu 20.04 desktop, and set a static IP address to 10.0.0.10 on the ethernet interface)
8. run `ping 10.0.0.2` from my desktop
9. When I do this, the board might successfully return a few ICMP messages, but it will inevitably start spitting out: `WARNING: Not destined for us; not forwardable... Dropping!`
10. When I actually probe this part of the code in ipv4input.c around line 262 where it checks that the IP address is correct, it tells me that the destination IP address is sometimes 10.0.0.10 (my desktop), sometimes 10.0.0.0 and sometimes 10.0.172.34 (not sure what this is).
11. When I check the traffic using wireshark, I can see that the destination IP address is always 10.0.0.2 (nuttx board), so I can only imagine that the networking stack is doing something wrong here.

**_Issue 2 - tcpecho example waiting for ACK after sending [SYN, ACK]_**

Hardware: Same hardware - STM32F767ZI nucleo development board
What I'm doing:
1. Cloned fresh nuttx and apps
2. Ran `./tools/configure.sh nucleo-144:f767-netnsh`
3. Used `make menuconfig` to enable network debugging and enable tcpecho example
4. Ran `make`
5. Flashed to board using openocd
6. Run minicom and reset board
7. Check default IP address (10.0.0.2 is the static of the nucleo-144, so I connected the board directly to my Ubuntu 20.04 desktop, and set a static IP address to 10.0.0.10 on the ethernet interface)
8. Inside NSH, run tcpecho
9. run `netcat 10.0.0.2 80` to connect to the default port 80 of tcpecho
10. The debug output spits out a whole bunch of stuff, including: `WARNING: Listen canceled while waiting for ACK on port 80`
11. When I check it with wireshark, it says that 10.0.0.2 sends [SYN, ACK] to 10.0.0.10, and then 10.0.0.10 sends an ACK back, but the nuttx board obviously doesn't get it. Then, the nuttx board sends [RST, ACK]

Output files:
[Issue1Output.txt](https://github.com/apache/incubator-nuttx/files/6247901/Issue1Output.txt)
[Issue2Output.txt](https://github.com/apache/incubator-nuttx/files/6247902/Issue2Output.txt)

I'm not sure if these are bugs, or if I just haven't configured the board properly. Any help would be greatly appreciated!

HecticSalad, 2021-04-05:
I managed to resolve issue 2 by enabling TCP/IP backlog support. 

Issue 1 is still outstanding.

acassis, 2021-04-05:
Hi @HecticSalad could you please algo attach your board config file? (the .config file)

HecticSalad, 2021-04-06:
Hey @acassis  sorry for the late reply. I have attached my config file in my initial message.

Thankyou!