apache/nuttx #3291: Standardize newline handling of syslog

v01d, 2021-04-03:
When looking into #3157 I realized we should make `syslog()` behave as hinted in the specification and as other implementations do. From #3157:

> Related to this discussion is the fact that I think syslog() should probably handle the newline adding on its own, as per the opengroup specification suggests (last phrase):
> 
>     The message body is generated from the message and following arguments in the same manner as if these were arguments to printf(), except that the additional conversion specification %m shall be recognized; it shall convert no arguments, shall cause the output of the error message string associated with the value of errno on entry to syslog(), and may be mixed with argument specifications of the "%n$" form. If a complete conversion specification with the m conversion specifier character is not just %m, the behavior is undefined. A trailing may be added if needed.
> 
> Linux has this in its manpage:
> 
>     The remaining arguments are a format, as in printf(3), and any arguments required by the format, except that
>     the two-character sequence %m will be replaced by the error message string strerror(errno). The format string
>     need not include a terminating newline character.
> 
> And this is BSD:
> 
>     The message is identical to a printf(3) format string, except that `%m'
>     is replaced by the current error message. (As denoted by the global
>     variable errno; see strerror(3).) A trailing newline is added if none is
>     present.
>     Newlines and other non-printable characters embedded in the message
>     string are printed in an alternate format. This prevents someone from
>     using non-printable characters to construct misleading log messages in an
>     output file. Newlines are printed as "\n", tabs are printed as "\t".
>     Other control characters are printed using a caret ("^") representation,
>     for example "^M" for carriage return.
> 

Thus, I think we should:
- [ ] make syslog append newline if not present in message
- [ ] replace other newlines with a printable character
- [ ] Remove all use of "\n" inside syslog calls (and correct any incoming PRs doing this)

hartmannathan, 2021-06-06:
Good candidate for "Good first issue" label?

fjpanag, 2023-07-10:
PR #8004 (which closed this issue), only addresses 1 of the 3 tasks listed.

I propose to re-open this, till all 3 tasks are resolved.

xiaoxiang781216, 2023-07-11:
Yes, let me repoen it.