apache/nuttx #3327: CHILD PTHREAD TERMINATION

v01d, 2021-04-05:
```
  Description: When a tasks exits, shouldn't all of its child pthreads also be
               terminated?

               This behavior was implemented as an options controlled by the
               configuration setting CONFIG_SCHED_EXIT_KILL_CHILDREN.  This
               option must be used with caution, however.  It  should not be
               used unless you are certain of what you are doing.  Uninformed
               of this option can often lead to memory leaks since, for
               example, memory allocations held by threads are not
               automatically freed!

  Status:      Closed.  No, this behavior will not be implemented unless
               specifically selected.
  Priority:    Medium, required for good emulation of process/pthread model.
               The current behavior allows for the main thread of a task to
               exit() and any child pthreads will persist.  That does raise
               some issues:  The main thread is treated much like just-another-
               pthread but must follow the semantics of a task or a process.
               That results in some inconsistencies (for example, with robust
               mutexes, what should happen if the main thread exits while
               holding a mutex?)
```

closed 2021-04-05 by v01d

xiaoxiang781216, 2021-04-23:
@v01d this content could be moved to FAQ?

fjpanag, 2021-04-23:
I don't see the option `CONFIG_SCHED_EXIT_KILL_CHILDREN` defined anywhere...

Is the original issue still applicable? Or has anything changed?

xiaoxiang781216, 2021-04-23:
Oh, it's the default behaviour and can't be disabled.