apache/nuttx #3330: GET_ENVIRON_PTR()

v01d, 2021-04-05:
```
  Description: get_environ_ptr() (sched/sched_getenvironptr.c) is not implemented.
               The representation of the environment strings selected for
               NuttX is not compatible with the operation.  Some significant
               re-design would be required to implement this function and that
               effort is thought to be not worth the result.
  Status:      Open.  No change is planned.
  Priority:    Low -- There is no plan to implement this.
```

xiaoxiang781216, 2021-10-31:
https://github.com/apache/incubator-nuttx/issues/3391
```
Description: The definition of environ in stdlib.h is bogus and will not
               work as it should.  This is because the underlying
               representation of the environment is not an array of pointers.
  Status:      Open
  Priority:    Medium
```

xiaoxiang781216, 2022-05-15:
Fix by https://github.com/apache/incubator-nuttx/pull/6083

closed 2022-05-15 by xiaoxiang781216