apache/nuttx #3334: execv() AND vfork()

v01d, 2021-04-05:
```
  Description: There is a problem when vfork() calls execv() (or execl()) to
               start a new application:  When the parent thread calls vfork()
               it receives and gets the pid of the vforked task, and *not*
               the pid of the desired execv'ed application.

               The same tasking arrangement is used by the standard function
               posix_spawn().  However, posix_spawn uses the non-standard, internal
               NuttX interface task_reparent() to replace the child's parent task
               with the caller of posix_spawn().  That cannot be done with vfork()
               because we don't know what vfork() is going to do.

               Any solution to this is either very difficult or impossible without
               an MMU.
  Status:      Open
  Priority:    Low (it might as well be low since it isn't going to be fixed).
```

yangyalei, 2023-03-07:
Really weird, I spend a hole day debug this.

yangyalei, 2023-03-08:
Now Implement：vforked process create a new task(https://github.com/apache/nuttx/blob/master/binfmt/binfmt_execmodule.c#L271) and exits(https://github.com/apache/nuttx/blob/master/sched/task/task_execve.c#L135)

Can wo and a new exec_module function that run cmd in current process? Looks like It doesn't matter with MMU.

xiaoxiang781216, 2023-03-08:
let's try it

closed 2023-09-14 by xiaoxiang781216