apache/nuttx #3392: TERMIOS

v01d, 2021-04-05:
```
  Description: Need some minimal termios support... at a minimum, enough to
               switch between raw and "normal" modes to support behavior like
               that needed for readline().
               UPDATE:  There is growing functionality in libs/libc/termios/
               and in the ioctl methods of several MCU serial drivers (stm32,
               lpc43, lpc17, pic32, and others).  However, as phrased, this
               bug cannot yet be closed since this "growing functionality"
               does not address all termios.h functionality and not all
               serial drivers support termios.
  Status:      Open
  Priority:    Low
```

acassis, 2021-04-05:
I think the next step is to list the Arch(es) that need(s) support. I noticed that the FreeModbus was modified to include "speed_t" that comes from Termios, this "inoffensive" modification was enough to make FreeModbus dependent on Termios. Now all MCUs will need to support Termios (or at like say it supports) to get FreeModbus compiled.

This modification is not a bad thing, modifying FreeModbus to use Termios will allow it to setup its own baudrate and parity without depending on user to configure it in Serial Driver Configuration inside the menuconfig.

robertlipe, 2021-05-04:
There's a LOT in "all termios.h functionality" that will never get used. Marking input parity errors (try that in a DMA/FIFO world), 50 baud, and FFDLY/FF1 (wait two seconds after sending a ^L without sending any data or exercising any flow control while the printer advances the platen since it "obviously" has no receive buffer) are a waste of time and complexity in this century. (And some of the previous century...) Full termios support would probably blow out a PIC.

I'd suggest narrowing this request substantially. Being able to support bit rate via integers instead of via CBAUD enums is nice and that's reasonable. If you didn't care about binary compatibility with ancient UNIX, you could do that with termio instead of termios.  Coding OLCUC and IUCLC because your five bit tty doesn't have lower case letters is a bit silly.

Is the bugreport here really "want readline to work" instead of "all termios functionality"?

Either Ken or Dennis called the terminal driver the biggest regret in original UNIX. :-)

patacongo, 2023-04-29:
I think that the basic functionality suggested by this Issue was added by several PRs in the Mar 1. 2023 timeframe mostly by @no1wudi   There is certainly more Termios development that will occur, but the support for basic RAW/COOKED mode operations have been implemented (ECHO, CR/LF, etc.).

closed 2023-04-29 by patacongo