apache/nuttx #3430: MODBUS NOT USABLE WITH USB SERIAL

v01d, 2021-04-05:
```
  Description: Modbus can be used with USB serial, however, if the USB
               serial connection is lost, Modbus will hang in an infinite
               loop.

               This is a problem in the handling of select() and read()
               and could probably resolved by studying the Modbus error
               handling.

               A more USB-friendly solution would be to: (1) Re-connect and
               (2) re-open the serial drivers.  That is what is done is NSH.
               When the serial USB device is removed, this terminates the
               session and NSH will then try to re-open the USB device.  See
               the function nsh_waitusbready() in the file
               apps/nshlib/nsh_usbconsole.c. When the USB serial is
               reconnected the open() in the function will succeed and a new
               session will be started.
  Status:      Open
  Priority:    Low.  This is really an enhancement request:  Modbus was never
               designed to work with removable serial devices.
```

fjpanag, 2021-04-25:
I think Modbus is only part of the apps repo.  
Maybe this is to be moved too?

acassis, 2021-04-25:
> I think Modbus is only part of the apps repo.
> Maybe this is to be moved too?

+1 -> apps