apache/nuttx #3516: CLOCKS_PER_SEC is not POSIX compliant

btashton, 2021-04-11:
NuttX defines `CLOCKS_PER_SEC` as:

```
#ifdef CONFIG_USEC_PER_TICK
# define CLK_TCK           (1000000/CONFIG_USEC_PER_TICK)
# define CLOCKS_PER_SEC    (1000000/CONFIG_USEC_PER_TICK)
#else
# define CLK_TCK           (100)
# define CLOCKS_PER_SEC    (100)
#endif
```
But the POSIX standard defines it as always one million.
https://pubs.opengroup.org/onlinepubs/9699919799/functions/clock.html

btashton, 2021-04-11:
@patacongo do you have any thoughts on this one?

patacongo, 2021-04-11:
> 
> 
> @patacongo do you have any thoughts on this one?

This implies that OpenGroup.org expects the system timer to ALWAYS have a resolution of 1 MSec.  Just changing the the setting to a million would break all architectures.  And, if all system timer APIs were modified to report / accept the time in "ticks" at a resolution of 1 MSec, then there were would be loss of functionality that is very important to an embedded system.

patacongo, 2021-04-11:
You are not the only one confused:  https://stackoverflow.com/questions/10455905/why-is-clocks-per-sec-not-the-actual-number-of-clocks-per-second

xiaoxiang781216, 2021-04-11:
From the thread:
```
The only thing the C99 N1256 standard draft says about CLOCKS_PER_SEC is that:
CLOCKS_PER_SEC which expands to an expression with type clock_t (described below) that is the number per second of the value returned by the clock function
```
And the quote from opengroup:
```
The functionality described on this reference page is aligned with the ISO C standard. Any conflict between the requirements described here and the ISO C standard is unintentional. This volume of POSIX.1-2017 defers to the ISO C standard. 
```
The value different from 1000000 is acceptable.

btashton, 2021-04-11:
Ah yes I see that the N1256 says nothing about this needing to be 1000000.  Unfortunate that the open posix test suite asserts this value.

patacongo, 2021-04-11:
> 
> 
> Ah yes I see that the N1256 says nothing about this needing to be 1000000. Unfortunate that the open posix test suite asserts this value.

Then it must also be a POSIX error that the return value of clock() is incorrect:  https://pubs.opengroup.org/onlinepubs/9699919799/functions/clock.html

patacongo, 2021-04-11:
I see you closed this.  Is that what we should do?  It appears to be a real POSIX non-conformance problem.

btashton, 2021-04-11:
My reading of the opengroup page is that it is OK if `CLOCKS_PER_SEC` is not 1M, as it is suppose to be the definition of C N1256 standard, it just usually is 1M.  What is important is that `clock()` / `CLOCKS_PER_SEC` is gives to process duration. Because we are using `clock_systime_ticks` this is not correct. 

My understanding is that we should to make clock() return the process ticks.  It would be _nice_ if we made CLOCKS_PER_SEC be 1M, but it would be ideal to just hide this in the clock() interfaces and use a real CLOCKS_PER_SEC in the OS internals.  This second bit I don't think is technically required.

casaroli, 2026-10-23:
The POSIX.1-2024 text, for reference.

[`<time.h>`](https://pubs.opengroup.org/onlinepubs/9799919799/basedefs/time.h.html):

> CLOCKS_PER_SEC
>
> A number used to convert the value returned by the clock() function into seconds.
> The value shall be an expression with type clock_t. [XSI] The value of CLOCKS_PER_SEC shall be 1 million on XSI-conformant systems. However, it may be variable on other systems, and it should not be assumed that CLOCKS_PER_SEC is a compile-time constant.

[`clock()`](https://pubs.opengroup.org/onlinepubs/9799919799/functions/clock.html):

> To determine the time in seconds, the value returned by clock() should be divided by the value of the macro CLOCKS_PER_SEC. [XSI] CLOCKS_PER_SEC is defined to be one million in <time.h>.

[`times()`](https://pubs.opengroup.org/onlinepubs/9799919799/functions/times.html):

> Applications should use sysconf(_SC_CLK_TCK) to determine the number of clock ticks per second as it may vary from system to system.

ISO C allows any value, but XSI requires 1000000. The system tick rate is a different value: `sysconf(_SC_CLK_TCK)`, the unit of `times()`. In NuttX, `CLK_TCK` already has this value.

The Open POSIX Test Suite checks the XSI value in [`conformance/interfaces/clock/2-1.c`](https://github.com/linux-test-project/ltp/blob/master/testcases/open_posix_testsuite/conformance/interfaces/clock/2-1.c) ("Test that CLOCKS_PER_SEC == 1,000,000 in <time.h>").

closed 2026-10-30 by xiaoxiang781216