apache/nuttx #3537: [cxd56xx] Not support hardware version v1.0B/v1.0C of WIFI board (iS110B)

fujitatomoya, 2021-04-13:
### Description

If use hardware version v1.0C of WIFI board (or iS110B), Wi-Fi driver reports error as following.
(About hardware version, refer to https://idy-design.com/product/is110b.html.)

```
up_assert: Assertion failed at file:wireless/gs2200m.c line: 1007 task: AppBringUp
up_registerdump: R0: 00000001 0d02bee0 000000e0 0d02e18c 0d02bee0 0d034248 0d02e178 0d0107c5
up_registerdump: R8: 0d026aa8 000003ef 0d033600 00000000 00000000 0d034248 0d002531 0d0027b0
up_registerdump: xPSR: 61000000 BASEPRI: 000000e0 CONTROL: 00000004
up_registerdump: EXC_RETURN: ffffffe9
up_dumpstate: sp:     0d034248
up_dumpstate: IRQ stack:
up_dumpstate:   base: 0d02b400
up_dumpstate:   size: 00000800
up_dumpstate:   used: 000000d0
up_dumpstate: User stack:
up_dumpstate:   base: 0d034380
up_dumpstate:   size: 000007e4
up_dumpstate:   used: 000002d0
up_stackdump: 0d034240: 0d033b9c 0d002675 000000e0 00000004 00000000 00000000 0d034248 0d002531
up_stackdump: 0d034260: 0d0027b0 00000006 000006e0 0d0107c5 00000000 0d0103e1 00000000 0d0268b0
up_stackdump: 0d034280: 000001a5 f8000600 a5a5a5a5 a5a5a5a5 0d038920 0d0342c0 00000000 0d02a704
up_stackdump: 0d0342a0: 0d0268b0 0d02a704 00000000 0d010915 0d038920 0d026928 0d0268b0 0d0113b3
up_stackdump: 0d0342c0: 30455441 04000a0d 041014a0 00000006 0d02cc70 0d034300 0d02cbf8 00000005
up_stackdump: 0d0342e0: 00000000 0d034300 0d02cbf8 00000005 00000000 0d00d403 00000000 00000005
up_stackdump: 0d034300: 97010104 99010104 9b010104 9d010105 00000000 00000000 0d02689d 00000000
up_stackdump: 0d034320: 00000000 00000000 00000000 0d00d035 00000000 01000000 00000000 00000000
up_stackdump: 0d034340: 00000000 43410000 7665642f 6d69742f 00317265 00000000 00000000 00000000
up_stackdump: 0d034360: 00000000 0d003549 00000001 0d034384 00000101 0d004417 00000000 00000000
up_taskdump: Idle Task: PID=0 Stack Used=0 of 0
up_taskdump: hpwork: PID=1 Stack Used=344 of 2028
up_taskdump: lpwork: PID=3 Stack Used=584 of 2028
up_taskdump: lpwork: PID=4 Stack Used=584 of 2028
up_taskdump: AppBringUp: PID=5 Stack Used=776 of 2020
up_taskdump: cxd56_pm_task: PID=6 Stack Used=352 of 996
```

### Reproducible Procedure

1. Plugin WIFI board v1.0C (iS110B)
2. tools/config.py feature/wifi
3. make
4. Flash spk
5. After rebooting, you will find above error message.

### Hardware

SPRESENSE Wi-Fi add-on board [iS110B](https://idy-design.com/product/is110b.html) version 1.0B and 1.0C

### Consideration

According to https://idy-design.com/product/is110b.html, there are 3 versions (v1.0A, v1.0B and v1.0C) of WIFI board for SPRESENSE main board.
For different hardware version, the connected pin is a little different.

After checking latest codes, find current codes are only for hardware version v1.0A.

https://github.com/apache/incubator-nuttx/blob/ff2ae3e894fa698d499090f676dc9e90f7611338/boards/arm/cxd56xx/common/src/cxd56_gs2200m.c#L224-L226

fujitatomoya, 2021-04-13:
original issue comes from https://github.com/sonydevworld/spresense/issues/137

fujitatomoya, 2021-04-13:
CC: @Barry-Xu-2018 @masayuki2009

fujitatomoya, 2021-04-13:
@Barry-Xu-2018

could you make PR based on https://github.com/Barry-Xu-2018/spresense-nuttx/commit/12135640469360c128456492a19efc887c2b2e23 to address this issue?

fujitatomoya, 2021-04-13:
FYI, i confirmed that v1.0B can be configured and working as expected with https://github.com/Barry-Xu-2018/spresense-nuttx/commit/12135640469360c128456492a19efc887c2b2e23

masayuki2009, 2021-04-13:
@fujitatomoya @Barry-Xu-2018

>SPRESENSE Wi-Fi add-on board iS110B version 1.0B and 1.0C

Thanks for reporting the issue.
To be honest, I did not notice that 1.0B and 1.0C were newly released.

> After checking latest codes, find current codes are only for hardware version v1.0A.

Yes, I only checked the driver with v1.0A

>FYI, i confirmed that v1.0B can be configured and working as expected with Barry-Xu-2018/spresense-nuttx@1213564

The code looks good to me.
So please send the PR to https://github.com/apache/incubator-nuttx/

Thanks.

Barry-Xu-2018, 2021-04-14:
@masayuki2009 @fujitatomoya 

I have created PR #3542.  
Please review it.  

thanks,

masayuki2009, 2021-04-14:
>I have created PR #3542.

The PR has been merged.
Let me close this issue.

closed 2021-04-14 by masayuki2009

fujitatomoya, 2021-04-14:
@masayuki2009 thanks for quick response, appreciate that 👍