apache/nuttx #3606: Add ccache option in build configuration.

fjpanag, 2021-04-25:
I recently so in the mailing list that NuttX is using ccache in CI builds, a tool that I was not aware of its existence (thank you!).  
It seems a nice tool that I have started using increasingly in my projects.

ccache supports two modes of operation:
1. It can be prefixed to the compiler commands.
2. It can "masquerade"  as the compiler, by using symlinks,.

It seems that NuttX build supports only the masquerade option.  
This is not optimal, as this forces you to have ccache always enabled for your whole development environment.

I am thinking of adding an option in build configuration settings (in Kconfig), so ccache may be used in prefix mode.  
This way it can be easily enabled/disabled, per-project, per-config, per-arch etc etc.

As I am not very experienced with this tool, please let me know if this is going to cause issues that I am not aware of, or if it affects in any way the CI.

v01d, 2021-04-25:
These symlinks are typically done on a separate path (/usr/lib/ccache in Ubuntu) so that you can choose to extend PATH with it or not. This is the same as when you decide to use a specific toolchain installed on a certain location.

That said, as I'm working on the CMake based build system it may be best to not invest in adding further complexity to the current build system, as this kind of thing is much easier in CMake.

fjpanag, 2021-04-26:
Sure this can wait till the cmake migration.

In any case though, I think that it is more user friendly to have this in Kconfig, compared to the alternatives...

closed 2022-11-04 by fjpanag