apache/nuttx #3612: RV32: syslog/printf handling of 64-bit integers is broken

gustavonihei, 2021-04-26:
The PR https://github.com/apache/incubator-nuttx/pull/3517 introduced a regression on the printing of 64-bit values through `printf` and `syslog`.

After a git bisect, it was found that the bad commit is https://github.com/apache/incubator-nuttx/pull/3517/commits/bff89f75153eaf09a5d377a90859d59daeab4e79.

It seems that this issue may to be related to the handling of variadic arguments. It could be reproduced on `esp32c3-devkit` and `bl602evb` targets. Other targets `sim` and `esp32-devkitc` are OK.

For reproducing the issue, execute the following application:
```c
#include <nuttx/config.h>
#include <stdint.h>
#include <inttypes.h>
#include <stdio.h>

int main(int argc, FAR char *argv[])
{
  uint64_t value = 0;
  printf("1=%" PRIx64 "\n", value);
  printf("2=%" PRIx64 "\n", value);
  printf("3=%" PRIx64 "\n", value);
  printf("4=%" PRIx64 "\n", value);

  return 0;
}
```
Expected output:
```
1=0
2=0
3=0
4=0
```
Actual output prints garbage values depending on current stack:
```
1=3fc820ac
2=3c010517
3=3c01051f
4=3c010527
```

davids5, 2021-04-26:
hmme we had this issues for ARM a long time back when the stack was not on an 8 byte alignment

gustavonihei, 2021-04-26:
I still haven't figured out why, but I am able to get the expected results by applying the following patch:
```patch
diff --git a/arch/risc-v/src/rv32im/riscv_initialstate.c b/arch/risc-v/src/rv32im/riscv_initialstate.c
index 4106125070..ab60becb1b 100644
--- a/arch/risc-v/src/rv32im/riscv_initialstate.c
+++ b/arch/risc-v/src/rv32im/riscv_initialstate.c
@@ -77,7 +77,7 @@ void up_initial_state(struct tcb_s *tcb)
    * only the start function would do that and we have control over that one
    */
 
-  xcp->regs[REG_SP]      = (uint32_t)tcb->stack_base_ptr +
+  xcp->regs[REG_SP]      = (uint32_t)tcb->stack_alloc_ptr +
                                      tcb->adj_stack_size;
 
   /* Save the task entry point */
```

xiaoxiang781216, 2021-04-27:
https://github.com/apache/incubator-nuttx/blob/master/arch/risc-v/src/common/riscv_createstack.c#L50-L54
```
#if defined(CONFIG_LIBC_FLOATINGPOINT) || defined (CONFIG_ARCH_RV64GC)
#  define STACK_ALIGNMENT   8
#else
#  define STACK_ALIGNMENT   4
#endif
```
Look like RISCV always need 8 bytes alignment?

xiaoxiang781216, 2021-04-27:
The same issue happen inside:
https://github.com/apache/incubator-nuttx/blob/master/arch/risc-v/src/common/riscv_usestack.c

gustavonihei, 2021-04-27:
> https://github.com/apache/incubator-nuttx/blob/master/arch/risc-v/src/common/riscv_createstack.c#L50-L54
>
> ```
> #if defined(CONFIG_LIBC_FLOATINGPOINT) || defined (CONFIG_ARCH_RV64GC)
> #  define [...]

Nice finding.
Actually, the [calling convention](https://github.com/riscv/riscv-elf-psabi-doc/blob/master/riscv-elf.md#user-content--integer-calling-convention:~:text=The%20stack%20grows%20downwards%20(towards%20lower,are%20stored%20at%20correspondingly%20higher%20addresses.) states the following:
```
The stack grows downwards (towards lower addresses) and the stack pointer shall be aligned to a 128-bit boundary upon procedure entry. 
```
So it seems that the correct `STACK_ALIGNMENT` should be 16.

saramonteiro, 2021-04-27:
Just more info to add:
The issue doesn't happen inside an ISR.

Ouss4, 2021-04-27:
> So it seems that the correct STACK_ALIGNMENT should be 16.

I've been wondering about this some time ago: https://github.com/apache/incubator-nuttx/pull/2769#issuecomment-769425201  it looks like it came back to bite us at the end..

There could still be few places in RISC-V port that came from MIPS and need cleaning.

gustavonihei, 2021-04-28:
#3614 fixed this issue. Closing it.

closed 2021-04-28 by gustavonihei