apache/nuttx #3663: Wi-Fi does not work on Photon board for nuttx-9.0.0 and newer releases

a-lunev, 2021-05-05:
First I tried to build the current state of the master branch and found that Wi-Fi does not work on Photon board.
Then I tested several older NuttX releases to check if some NuttX release normally works.
As it turned out, nuttx-8.2 works well, but nuttx-9.0.0 already does not. Thus Wi-Fi functionality of Photon board was broken in NuttX somewhere between nuttx-8.2 and nuttx-9.0.0.

Steps to reproduce:
<pre>
$ git clone https://github.com/apache/incubator-nuttx.git nuttx
$ git clone https://github.com/apache/incubator-nuttx-apps apps
$ cd nuttx
$ ./tools/configure.sh -l photon:wlan
$ make menuconfig
</pre>
Application Configuration → Network Utilities → Network initialization → WAPI Configuration
set SSID
set Passprhase
save, exit
<pre>
$ make CROSSDEV=arm-none-eabi-
$ sudo dfu-util -d 2b04:d006 -a 0 -s 0x08020000 -D nuttx.bin
</pre>

acassis, 2021-05-05:
Hi @a-lunev did you try to use "git bisect" to spot the bad commit that caused the issue?

a-lunev, 2021-05-06:
Hi Alan,

git bisecting nuttx:
$ git checkout nuttx-9.0.0
$ git bisect start
$ git bisect bad
$ git bisect good nuttx-8.2

good: nuttx: fd3e2c59168cbb8f34713c2cceaf01c1786e1a2a, apps: 8c9663b6c121a66189850662d7ce7e51649e078e
good: nuttx: 898141300c11bf06bc7a653b8651a744527a1e60, apps: 8c9663b6c121a66189850662d7ce7e51649e078e
good: nuttx: 55907b00626d13accab61a1cdb66d81a00e6a919, apps: 8c9663b6c121a66189850662d7ce7e51649e078e
good: nuttx: 72104c182c669e5ea1c8bdef2f6bfec273e4beee, apps: 8c9663b6c121a66189850662d7ce7e51649e078e
good: nuttx: 9b86305ad333e0bac2bddf4715f241b9996e1e3c, apps: 8c9663b6c121a66189850662d7ce7e51649e078e
good: nuttx: 8b4cfbbd8523b7dc5a0be3cacf8c606971843518, apps: 8c9663b6c121a66189850662d7ce7e51649e078e
good: nuttx: ca367a259b89a1d289e15f9fdbb6e16c2d41471e, apps: 8c9663b6c121a66189850662d7ce7e51649e078e
good: nuttx: 0335c5251e8a2981a9edc05dedba9ca058db1574, apps: 8c9663b6c121a66189850662d7ce7e51649e078e
good: nuttx: 932c2f3b6f6a44739c96128e7d995e43749dbb79, apps: 8c9663b6c121a66189850662d7ce7e51649e078e
good: nuttx: 3eaba41ab9212340ff9376f0c00da190db3f0dea, apps: 8c9663b6c121a66189850662d7ce7e51649e078e
good: nuttx: 725bdfb0e8c704823669d20931fd0a824c462212 (nuttx-9.0.0), apps: 8c9663b6c121a66189850662d7ce7e51649e078e

git bisecting apps:
$ git checkout nuttx-9.0.0
$ git bisect start
$ git bisect bad
$ git bisect good 8c9663b6c121a66189850662d7ce7e51649e078e

bad: apps: da31673ddfdab1a634093265c0e6b8c4da55e99d, nuttx: nuttx-9.0.0
good: apps: d660492289a0766846b59ab2587ae310e15b91fd, nuttx: nuttx-9.0.0
bad: apps: dd9059203daf12face37491f24c97fd4f2f3f03f, nuttx: nuttx-9.0.0
good: apps: b2e9b46fb1b567f5a03f8f53a133e6e0a7a914ad, nuttx: nuttx-9.0.0
bad: apps: 404b330c25567923de8434e34dd1dbe8ccf59b8b, nuttx: nuttx-9.0.0
bad: apps: 86bfe1339969606d1101d1d3d1d3d7bb07705bbe, nuttx: nuttx-9.0.0
good: apps: ee615f0568735f590441ca3dcbb8202ed0e271be, nuttx: nuttx-9.0.0

86bfe1339969606d1101d1d3d1d3d7bb07705bbe is the first bad commit (https://github.com/apache/incubator-nuttx-apps)
ee615f0568735f590441ca3dcbb8202ed0e271be is the last good commit (https://github.com/apache/incubator-nuttx-apps)

anchao, 2021-05-06:
Oops, seems regression caused by my pull request, I will look into this issue and update the process later.

a-lunev, 2021-05-06:
Hi @anchao, thank you!

anchao, 2021-05-06:
Hi @a-lunev ,

Thanks for your investigation, Please try this patch   https://github.com/apache/incubator-nuttx-apps/pull/702

acassis, 2021-05-06:
Nice work @a-lunev Kudos!!!

a-lunev, 2021-05-06:
@anchao, I've tested the patch. The issue is resolved. Thank you very much!

acassis, 2021-05-06:
Thank you @a-lunev ! I will close it!

closed 2021-05-06 by acassis