apache/nuttx #3686: low baud rate serial port not supported

acassis, 2021-05-09:
Currently many MCUs support by NuttX are not working at low baudrate.

Although nobody will want to use low baudrate for serial console, many industrial protocols work at low baudrate. For instance the ABNT CODI used on Energy Meter in Brazil works at 110bps.

Testing STM32L496 I noticed it doesn't work below 4800 bps, and STM32F407 used on STM32F4Discovery doesn't work below 1200 bps.

The easy way to test lower baud rate is configuring the NuttX Serial Console port to work with a value different of 115200.

Currently ESP32 is one of few MCUs to works fine even at 110 bps! After configuring the Serial Console to 110 bps (Device Drivers -> Serial Driver Support -> UART0 Configuration -> (110) BAUD rate) use picocom to test (minicom only went down to 300bps)
```
$ picocom -b 110 /dev/ttyUSB0
```
So supporting lower baud rate is needed to use NuttX for Energy Meter application here in Brazil.

acassis, 2021-05-10:
Hi @davids5 that is exactly what I did.
I noticed that configuring the STM32L496 to 110 bps was resulting a baud rate of 12547 bps, because the mask to 16-bit was removing the higher part. Then I changed the SYSCLK from 80MHz to 48MHz and divided it by 16 to get 3MHz: APB1 clock (PCLK1) is HCLK/16 (3MHz). Other modification was "#  define STM32L4_CLK48_SEL       RCC_CCIPR_CLK48SEL_PLLMAIN" to get USB and serial working at same time.

closed 2021-05-10 by acassis

davids5, 2021-05-10:
Did you try setting PSC? It is indicated for Irda/7816 but it may work as well.

acassis, 2021-05-10:
No, I didn't try using the PSC, the RM states it is for IrDA and SmartCard, but since the field exists on USART_GTPR for all USARTs, it could be an option at least to try.