apache/nuttx #3692: Basic IP Cores requirement for NuttX  for Zybo Z7 or any FPGA

bhatmahadev6, 2021-05-10:
Dear @btashton,

What are the IP cores should be added in Zybo Z7 platform which is already written in  litex-boards( https://github.com/litex-hub/litex-boards)

Does generating the bitstream with a UART and VexRISC-V core (which has D-cache and I-cache) suffice for booting NuttX.
(i am asking this because the Arty A7 build has extra IP Cores such as LiteDRAM,LiteEthernet etc..)

with the existing platform and doing make menuconfig would hopefully work right ?

btashton, 2021-05-10:
That's mostly correct. You will need to also figure out what you want to use for RAM and update the linker script accordingly. The Arty A7 uses the DDR for both the program (8M) and the RAM (256-8M).

bhatmahadev6, 2021-05-18:
HI Brennan Ashton,
How to get this for the Zybo ?

The below one is for Arty.

litex-boards/platform/digilent_zybo.py
    # DDR3 SDRAM
    ("ddram", 0,
        Subsignal("a", Pins(
            "R2 M6 N4 T1 N6 R7 V6 U7",
            "R8 V7 R6 U6 T6 T8"),
            IOStandard("SSTL135")),
        Subsignal("ba",    Pins("R1 P4 P2"), IOStandard("SSTL135")),
        Subsignal("ras_n", Pins("P3"), IOStandard("SSTL135")),
        Subsignal("cas_n", Pins("M4"), IOStandard("SSTL135")),
        Subsignal("we_n",  Pins("P5"), IOStandard("SSTL135")),
        Subsignal("cs_n",  Pins("U8"), IOStandard("SSTL135")),
        Subsignal("dm", Pins("L1 U1"), IOStandard("SSTL135")),
        Subsignal("dq", Pins(
            "K5 L3 K3 L6 M3 M1 L4 M2",
            "V4 T5 U4 V5 V1 T3 U3 R3"),
            IOStandard("SSTL135"),
            Misc("IN_TERM=UNTUNED_SPLIT_40")),
        Subsignal("dqs_p", Pins("N2 U2"),
            IOStandard("DIFF_SSTL135"),
            Misc("IN_TERM=UNTUNED_SPLIT_40")),
        Subsignal("dqs_n", Pins("N1 V2"),
            IOStandard("DIFF_SSTL135"),
            Misc("IN_TERM=UNTUNED_SPLIT_40")),
        Subsignal("clk_p", Pins("U9"), IOStandard("DIFF_SSTL135")),
        Subsignal("clk_n", Pins("V9"), IOStandard("DIFF_SSTL135")),
        Subsignal("cke",   Pins("N5"), IOStandard("SSTL135")),
        Subsignal("odt",   Pins("R5"), IOStandard("SSTL135")),
        Subsignal("reset_n", Pins("K6"), IOStandard("SSTL135")),
        Misc("SLEW=FAST"),
    ),

In litex-boards/targets/digilent_arty.py
        # DDR3 SDRAM -------------------------------------------------------------------------------
        if not self.integrated_main_ram_size:
            self.submodules.ddrphy = s7ddrphy.A7DDRPHY(platform.request("ddram"),
                memtype        = "DDR3",
                nphases        = 4,
                sys_clk_freq   = sys_clk_freq)
            self.add_sdram("sdram",
                phy           = self.ddrphy,
                module        = MT41K128M16(sys_clk_freq, "1:4"),
                l2_cache_size = kwargs.get("l2_size", 8192)
            )
we just need changes for these 2 right ?

Thanks

bhatmahadev6, 2021-05-18:
For Arty it was already there from Xilinx with the appropriate Inout standards in their page.(mig.prj)

But I am exactly not understanding how we can do it for Zybo.(I know it can be done with MIG Core but please tell me the appropriate steps)