apache/nuttx #3711: Raspberry Pi Pico SPI SD Card `ERROR: Failed to mount the SDCARD. -19`

microhobby, 2021-05-13:
Trying to test the SD Card support for RPI Pico configuring the build with `./tools/configure.sh -l raspberrypi-pico:spisd` (and adding the debug features)  I'm not having success. Driver initialization appears to be successful but the mount returns ENODEV:

```
board_spisd_initialize: Initializing SPI for the MMC/SD slot
mmcsd_mediainitialize: Send CMD0
mmcsd_waitready: Card still busy, last response: 00
mmcsd_sendcmd: CMD0[00000000] R1=01
mmcsd_mediainitialize: Card is in IDLE state
mmcsd_mediainitialize: Send CMD8
mmcsd_sendcmd: CMD8[000001aa] R1=01 R7=000001aa
mmcsd_mediainitialize: 268584768. Send CMD55/ACMD41
mmcsd_sendcmd: CMD55[00000000] R1=01
mmcsd_sendcmd: CMD41[40000000] R1=01
mmcsd_mediainitialize: 53. Send CMD55/ACMD41
mmcsd_sendcmd: CMD55[00000000] R1=01
mmcsd_sendcmd: CMD41[40000000] R1=01
mmcsd_mediainitialize: 53. Send CMD55/ACMD41
mmcsd_sendcmd: CMD55[00000000] R1=01
mmcsd_sendcmd: CMD41[40000000] R1=01
mmcsd_mediainitialize: 53. Send CMD55/ACMD41
mmcsd_sendcmd: CMD55[00000000] R1=01
mmcsd_sendcmd: CMD41[40000000] R1=01
mmcsd_mediainitialize: 53. Send CMD55/ACMD41
mmcsd_sendcmd: CMD55[00000000] R1=01
mmcsd_sendcmd: CMD41[40000000] R1=01
mmcsd_mediainitialize: 53. Send CMD55/ACMD41
mmcsd_sendcmd: CMD55[00000000] R1=01
mmcsd_sendcmd: CMD41[40000000] R1=01
mmcsd_mediainitialize: 53. Send CMD55/ACMD41
mmcsd_sendcmd: CMD55[00000000] R1=01
mmcsd_sendcmd: CMD41[40000000] R1=01
mmcsd_mediainitialize: 53. Send CMD55/ACMD41
mmcsd_sendcmd: CMD55[00000000] R1=01
mmcsd_sendcmd: CMD41[40000000] R1=01
mmcsd_mediainitialize: 53. Send CMD55/ACMD41
mmcsd_sendcmd: CMD55[00000000] R1=01
mmcsd_sendcmd: CMD41[40000000] R1=01
mmcsd_mediainitialize: 53. Send CMD55/ACMD41
mmcsd_sendcmd: CMD55[00000000] R1=01
mmcsd_sendcmd: CMD41[40000000] R1=01
mmcsd_mediainitialize: 53. Send CMD55/ACMD41
mmcsd_sendcmd: CMD55[00000000] R1=01
mmcsd_sendcmd: CMD41[40000000] R1=01
mmcsd_mediainitialize: 53. Send CMD55/ACMD41
mmcsd_sendcmd: CMD55[00000000] R1=01
mmcsd_sendcmd: CMD41[40000000] R1=01
mmcsd_mediainitialize: 53. Send CMD55/ACMD41
mmcsd_sendcmd: CMD55[00000000] R1=01
mmcsd_sendcmd: CMD41[40000000] R1=01
mmcsd_mediainitialize: 53. Send CMD55/ACMD41
mmcsd_sendcmd: CMD55[00000000] R1=01
mmcsd_sendcmd: CMD41[40000000] R1=01
mmcsd_mediainitialize: 53. Send CMD55/ACMD41
mmcsd_sendcmd: CMD55[00000000] R1=01
mmcsd_sendcmd: CMD41[40000000] R1=01
mmcsd_mediainitialize: 53. Send CMD55/ACMD41
mmcsd_sendcmd: CMD55[00000000] R1=01
mmcsd_sendcmd: CMD41[40000000] R1=01
mmcsd_mediainitialize: 53. Send CMD55/ACMD41
mmcsd_sendcmd: CMD55[00000000] R1=01
mmcsd_sendcmd: CMD41[40000000] R1=01
mmcsd_mediainitialize: 53. Send CMD55/ACMD41
mmcsd_sendcmd: CMD55[00000000] R1=01
mmcsd_sendcmd: CMD41[40000000] R1=01
mmcsd_mediainitialize: 53. Send CMD55/ACMD41
mmcsd_sendcmd: CMD55[00000000] R1=01
mmcsd_sendcmd: CMD41[40000000] R1=01
mmcsd_mediainitialize: 53. Send CMD55/ACMD41
mmcsd_sendcmd: CMD55[00000000] R1=01
mmcsd_sendcmd: CMD41[40000000] R1=01
[... 163 lines cut ...]
mmcsd_dmpcsd:   COPY:                    0
mmcsd_dmpcsd:   PERM_WRITE_PROTECT:      0
mmcsd_dmpcsd:   TMP_WRITE_PROTECT:       0
mmcsd_dmpcsd:   FILE_FORMAT:             0
mmcsd_dmpcsd:   CRC:                     65
mmcsd_decodecsd: SPI Frequency
mmcsd_decodecsd:   Maximum:         25000000 Hz
mmcsd_decodecsd:   Actual:          20833333 Hz
mmcsd_decodecsd: Read access time:  11 ticks
mmcsd_decodecsd: Write access time: 26 ticks
mmcsd_decodecsd: Sector size:       512
mmcsd_decodecsd: Number of sectors: 15593472
mmcsd_spislotinitialize: mmcsd_mediainitialize returned OK
find_blockdriver: pathname="/dev/mmcsd0"
mmcsd_open: Entry
mmcsd_sendcmd: CMD9[00000000] R1=00
mmcsd_getcardinfo: 0. SPI send returned ff
mmcsd_getcardinfo: 1. SPI send returned ff
mmcsd_getcardinfo: 2. SPI send returned 20
mmcsd_getcardinfo: 3. SPI send returned 07
mmcsd_getcardinfo: ERROR: 3. Data transfer error: 07
mmcsd_geometry: ERROR: mmcsd_getcsd returned -5
nx_mount: ERROR: Bind method failed: -19
board_spisd_initialize: ERROR: Failed to mount the SDCARD. -19
```

acassis, 2021-05-13:
Hi @microhobby it is failing to get card information during the mount process, it is failing here:
```
  for (i = 0; i < 8; i++)
    {
      response = SPI_SEND(spi, 0xff);
      finfo("%d. SPI send returned %02x\n", i, response);

      /* If a read operation fails and the card cannot provide the requested
       * data, it will send a data error token instead.  The 4 least
       * significant bits are the same as those in the R2 response.
       */

      if (response != 0 && (response & MMCSD_SPIDET_UPPER) == 0)
        {
          ferr("ERROR: %d. Data transfer error: %02x\n", i, response);
          return -EIO;
        }
```
Did you try with more SDCards ?
Also, how are you wiring the module? I noticed it is running at relative high frequency for SPI (20MHz) :
```
mmcsd_decodecsd:   Maximum:         25000000 Hz
mmcsd_decodecsd:   Actual:          20833333 Hz
```
Maybe using short wires could help!

microhobby, 2021-05-13:
Thanks for the help @acassis , you're right I'm using "long" wires. I will solder another header on my module to put it direct on the protoboard for my tests.

patacongo, 2021-05-13:
> Also, how are you wiring the module? I noticed it is running at 
> relative high frequency for SPI (20MHz) :
>
> |mmcsd_decodecsd: Maximum: 25000000 Hz mmcsd_decodecsd: Actual: 
> 20833333 Hz |
>
> Maybe using short wires could help!
>
Hmm.. I think that the maximum frequency for a standard SD card in SPI 
mode is 20MHz, is it not?  20.8MHz is slightly over-clocked (but might 
be fine for some higher speed cards).

The SD card initialization is done in ID mode which is 400KHz. After 
that, the operational mode uses a max of 20MHz.  So when you mount the 
SD card, this will be first transaction in 20MHz operational mode.  So a 
frequency issue could well be the problem.

patacongo, 2021-05-13:
On 5/13/2021 2:29 PM, Gregory Nutt wrote:
> Hmm.. I think that the maximum frequency for a standard SD card in SPI
> mode is 20MHz, is it not?  20.8MHz is slightly over-clocked (but might
> be fine for some higher speed cards).
>
> The SD card [...]
No... I checked.  I was confused.  25MHz is the correct maximum 
frequency from both 4-bit and SPI modes.

microhobby, 2021-05-14:
@acassis  @patacongo I was using the follow to connect:

[image]

Now I switched to rigid jumpers:

[image]

But I still have the same issue. I tested with three different SD cards.

microhobby, 2021-05-14:
@acassis @patacongo I check the `CONFIG_MMCSD_SPICLOCK` and it is set by default to `20000000`. If I change `CONFIG_MMCSD_SPICLOCK` to `400000` (like the ID mode clock) it works

patacongo, 2021-05-14:
> 
> 
> @acassis @patacongo I check the `CONFIG_MMCSD_SPICLOCK` and it is set by default to `20000000`. If I change `CONFIG_MMCSD_SPICLOCK` to `400000` (like the ID mode clock) it works

It is not unusual to have to tune the SD clock in cases like this.  You can experiment and find an optimal frequency for your board configuration.  If you go to PCB, you will likely have to retune that clock.

I would assume this is an RF noise issue.  Did you try using shorter wires like Alan suggested?

microhobby, 2021-05-14:
> It is not unusual to have to tune the SD clock in cases like this. You can experiment and find an optimal frequency for your board configuration. If you go to PCB, you will likely have to retune [...]

Yeah, I'm using short rigid jumper wires now https://github.com/apache/incubator-nuttx/issues/3711#issuecomment-840920928

microhobby, 2021-05-14:
Thanks @acassis and @patacongo , the tips here helped me a lot. I also believe that it can be something in my environment, but now I can continue with my tests so I'm going to close this ticket.

closed 2021-05-14 by microhobby

linguini1, 2024-10-25:
Just wanted to comment for visibility that I was having the same issue and reducing clock speed helped me resolve it. Maybe I could include this on the Pico doc page for the spisd config as a note? I imagine most people are using SD adapters with a similar bread board setup.