apache/nuttx #3751: nucleo-h743zi:elf config got broken (nsh: nsh_session: readline failed: 13)

a-lunev, 2021-05-20:
Steps to reproduce:

```
$ mkdir TEST_ROOT
$ git clone https://github.com/apache/incubator-nuttx.git TEST_ROOT/nuttx
$ git clone https://github.com/apache/incubator-nuttx-apps TEST_ROOT/apps
$ cd TEST_ROOT/nuttx
$ ./tools/configure.sh -l nucleo-h743zi:elf
$ make CROSSDEV=arm-none-eabi-
$ sudo openocd -f interface/stlink.cfg \
               -f board/st_nucleo_h743zi.cfg \
               -c "init" \
               -c "reset init" \
               -c "flash probe 0" \
               -c "flash info 0" \
               -c "flash write_image erase nuttx.bin 0x08000000" \
               -c "reset run" -c "shutdown"
```

```
nsh> ls
nsh> nsh: nsh_session: readline failed: 13
```

The last known working pair was:
nuttx: 705c64e5ff50df4f7ea202ad870e7106dddd243c (13-Dec-2020)
apps: f2100104663704be498179f06bd78adcbe62a809 (17-Dec-2020)

Thus the issue has appeared somewhere between 13-Dec-2020 and now.

a-lunev, 2021-05-20:
git bisecting nuttx:
$ git checkout e4fd99682e4c061dcdc4db7751d1a12c19d96eba
$ git bisect start
$ git bisect bad
$ git bisect good 705c64e5ff50df4f7ea202ad870e7106dddd243c

good: nuttx: f16a0a7380d64a01dc9cf4839f5b2d5e611f1f99, apps: f2100104663704be498179f06bd78adcbe62a809
good: nuttx: d1e06b4c0919e49f9589a295f66dbd50db8961a3, apps: f2100104663704be498179f06bd78adcbe62a809
good: nuttx: 7fb73dba1c2175d720170f21094194130819ff81, apps: f2100104663704be498179f06bd78adcbe62a809
good: nuttx: 649e680f75b3a247be98e8624e03b3c7fe0ad600, apps: f2100104663704be498179f06bd78adcbe62a809
bad: nuttx: 09a0ed111c0c193a57b2a538e23c2b6a6e9cdf26, apps: f2100104663704be498179f06bd78adcbe62a809
bad: nuttx: 1856a85b2850a49065e8869ccc952fb98dd02c34, apps: f2100104663704be498179f06bd78adcbe62a809
bad: nuttx: db517cc2293ea53bebd9730d20ed35db46c08b3b, apps: f2100104663704be498179f06bd78adcbe62a809
bad: nuttx: 90a4e8d718c269ac110289ea0fac368abe78ef54, apps: f2100104663704be498179f06bd78adcbe62a809
bad: nuttx: bf733f72132d387206c3ffe21fb6b2a085dd5506, apps: f2100104663704be498179f06bd78adcbe62a809
bad: nuttx: 468e08c1b4ff1441cf898623782869c7daf1c947, apps: f2100104663704be498179f06bd78adcbe62a809
good: nuttx: e5f10696541dcbd53337b90f53538a48d073b4bc, apps: f2100104663704be498179f06bd78adcbe62a809

468e08c1b4ff1441cf898623782869c7daf1c947 is the first bad commit
e5f10696541dcbd53337b90f53538a48d073b4bc is the last good commit

patacongo, 2021-05-20:
> [468e08c](https://github.com/apache/incubator-nuttx/commit/468e08c1b4ff1441cf898623782869c7daf1c947) is the first bad commit
> [e5f1069](https://github.com/apache/incubator-nuttx/commit/e5f10696541dcbd53337b90f53538a48d073b4bc) is the last good commit

When I first looked, 468e08c seemed like an unlikely cause of the problem.  That commit simply changes some error handling if the ELF load fails.  The ELF loader simply calls `elf_uninit()` in the event of a failure:

    65 int elf_uninit(struct elf_loadinfo_s *loadinfo)
    66 {
    67   /* Free all working buffers */
    68
    69   elf_freebuffers(loadinfo);
    70
    71   /* Close the ELF file */
    72
    73   if (loadinfo->filfd >= 0)
    74     {
    75       nx_close(loadinfo->filfd);
    76     }
    77
    78   return OK;
    79 }

You see a failure that is attributed to readline() which seems unrelated:

    79 #define EACCES              13
    80 #define EACCES_STR          "Permission denied"

The error is reported, apparently, by logic in `apps/nshlib/nsh_stdsession.c` if a call to `readline()` returns EOF.  `readline()` ultimately calls `read()` on stdin (file descriptor 0).  `EACCES` is returned from `read()` if the file is not opened for reading:

    71   /* Was this file opened for read access? */
    72
    73   if ((filep->f_oflags & O_RDOK) == 0)
    74     {
    75       /* No.. File is not read-able */
    76
    77       ret = -EACCES;
    78     }
    79

So now I think I see how this commit can cause this error to be seen:

1. When ELF support is enabled, NSH will always first attempt to a program with that command name at the PATH directory.  For example, `nsh> ls` will attempt to run at `/bin/ls`
2. In this case it will fail to run the problem and will fail silently.  It will call `elf_unit()` which will close the file at descriptor `loadinfo->filfd`.  But that file descriptor has not been initialized and is still at its initial state; of zero:

    elf_init():
    136   memset(loadinfo, 0, sizeof(struct elf_loadinfo_s));

3. So `stdin` will be closed (via close -> nx_close -> file_close).
4. After this fails, then readline() is called which will attempt to `read()` from `stdin`. But `read()` will fail will errno `EACCES` because stdin is no open for writing.

a-lunev, 2021-05-20:
Hi @patacongo,

Thank you for your investigation.
I tried your patch (loadinfo->filfd instead of loadinfo->fildfd). Yes, it resolves the issue.

The corrected patch:
```
diff --git a/binfmt/libelf/libelf_init.c b/binfmt/libelf/libelf_init.c
index 6658aa0b62..f2da035e41 100644
--- a/binfmt/libelf/libelf_init.c
+++ b/binfmt/libelf/libelf_init.c
@@ -134,6 +134,7 @@ int elf_init(FAR const char *filename, FAR struct elf_loadinfo_s *loadinfo)
   /* Clear the load info structure */
 
   memset(loadinfo, 0, sizeof(struct elf_loadinfo_s));
+  loadinfo->filfd = -1;
 
   /* Get the length of the file. */
```

patacongo, 2021-05-20:
> I tried your patch (loadinfo->filfd instead of loadinfo->fildfd). Yes, it resolves the issue.

Can you submit a PR?

closed 2021-05-21 by davids5