apache/nuttx #3765: GSoC: NuttX support for pysimCoder (info thread)

michallenc, 2021-05-23:
Dear NuttX community,

as was already mentioned in the mailing list, my project NuttX support for pysimCoder ([proposal can be found here](https://docs.google.com/document/d/13jipDxufiD3QGfJ0Udnp_B_3JxieoPO3lvrCUvLVNhQ/edit)) has been accepted for Google Summer of Code under Apache Foundation. I would like to express my gratitude to Apache and NuttX admins for this opportunity as well as to my project mentors Roberto Bucher, Pavel Píša and Brennan Ashton. We are currently discussing details about the implementations as part of a pre coding period.

I will use this GitHub issue for providing regular updates about the project progress once the coding period starts on 7th June. We can also use this thread to discuss some further ideas regarding the NuttX support for pysimCoder. The integration has a great potencial to allow more comfortable NuttX usage in control systems.

michallenc, 2021-06-11:
Hello all,

the first week of GSoC coding period is about to end and I´d like to refer about the progress we made during the first week and during pre coding period.

During pre GSoC period I extended pysimCoder CAN support for NuttX with SocketCAN support so pysimCoder now supports both characteristic device dev/can0 and socketcan. This was [integrated into pysimCoder mainline last week](https://github.com/robertobucher/pysimCoder/pull/6). This week was mainly about incremental encoder and DC motor control, which has been done on STM32 Nucleo F446RE and presented to the mentors during our weekly meeting. DC motor control required support for GPIO driver for Nucleo F446RE which was implemented to NuttX in [PR 3880 ](https://github.com/apache/incubator-nuttx/pull/3880). This implementation also required [new pysimCoder block for relational operation](https://github.com/robertobucher/pysimCoder/commit/5c7c49188018f9292e2e005f2c62f3ab471049e3).

The main platform of the project is iMXRT 1060 MCU so the goal is also to get the DC motor control running there, but I am currently having some troubles with encoder driver for imxrt. I already did some fixes to xbar driver (added missing clock), but still some work is to be done in encoder driver (as it stands now, the driver registers the device but do not update the position value).

Yesterday I created a wiki page about the project on pysimCoder git. We agreed during the meeting that the goal would be to store links to commits to pysimCoder and NuttX there as well as the content of our meetings. [The page can be found here](https://github.com/robertobucher/pysimCoder/wiki/GSoC-2021:-pysimCoder-integration-with-NuttX).

michallenc, 2021-06-20:
Hello,

another week of GSoC is behind me so I´d like to post some updates since the last week. I extended project wiki from pysimCoder github to NuttX wiki subfolder few days ago, [which can be found here](https://cwiki.apache.org/confluence/display/NUTTX/%5B2021%5D+NuttX+Support+for+Rapid+Control+Applications+Development+with+pysimCoder). Proper documentation containing information about compiling NuttX with pysimCoder then will be created under Documentation by the end of GSoC period.

The problem with the encoder driver for iMXRT MCU was succesfully solved thanks to my mentor dr. Píša´s help during the debugging session and [the changes have been merge to NuttX mainline](https://github.com/apache/incubator-nuttx/pull/3929). I mainly focused on the support for sensors and DAC for pysimCoder then. [Support for DHTXX sensor was added as an sensors example yesterday](https://github.com/robertobucher/pysimCoder/pull/11) and [block for DAC driver was also tested with Nucleo board](https://github.com/robertobucher/pysimCoder/pull/12) (iMXRT does not support DAC for external signals).

acassis, 2021-06-20:
Hi @michallenc thank you for reporting the status and the issues you faced. About the DAC are you planing to create a driver to iMXRT or just testing on Nucleo board is enough?

michallenc, 2021-06-21:
Hi @acassis. Unfortunately there is no digital/analog converter output on the iMXRT 1060 chip at all. There is just a 6 bit DAC for on chip internal signals only, but it´s not routed to the external pins (chapter 65 in the reference manual). So DAC can be tested just on Nucleo board.

DAC support should be in newer series iMXRT11xx chips though, but those are not supported by NuttX yet.

acassis, 2021-06-21:
Hmm, understood to point. In fact it is sad they didn't included the DAC support in the board!

michallenc, 2021-06-28:
Hello,

last week of GSoC was mainly about multiple channel PWM support in pysimCoder and PWM support for iMXRT (this was originally by the end of GSoC but I decided to create the driver now so it can be properly tested on iMXRT instead of just STM32). Both [multiple channel support in pysimCoder](https://github.com/robertobucher/pysimCoder/pull/13) and [PWM driver for iMXRT](https://github.com/apache/incubator-nuttx/pull/3980) where merged to pysimCoder and NuttX mainline, respectively. I plan to create a single block in pysimCoder which would be able to control DC motor connected to the board with NuttX this week. Current sensing application is also planned.

Last week I also forgot to mention that I had applied for 2021 NuttX Worshop with my presentation about pysimCoder integration with NuttX, so I am looking forward to that, if my presentation gets to the workshop ofcourse.

michallenc, 2021-07-03:
Hello,

during this weekend, I´ve dedicated most of my time into putting into practice my lasts commits regarding the changes to iMXRT driver and also getting running the TCP sender to a real time plotter. The result of the effort is a [short demo of DC motor control](https://www.youtube.com/watch?v=6HlGk3ecPNQ) uploaded by @acassis on NuttX YouTube channel. List of the new commits to pysimCoder that introduced TCP block, PID control and some other smaller changes can be found on [NuttX wiki](https://cwiki.apache.org/confluence/display/NUTTX/%5B2021%5D+NuttX+Support+for+Rapid+Control+Applications+Development+with+pysimCoder) or [pysimCoder wiki](https://github.com/robertobucher/pysimCoder/wiki/GSoC-2021:-pysimCoder-integration-with-NuttX).

I had to make some changes into serRTScope.py programm in pysimCoder library so the plotter can support TCP data, [the result can be found in my fork repository](https://github.com/michallenc/pysimCoder/blob/dc_motor/BlockEditor/tcpRTScope.py). There are still some changes to do, mainly on the graphic side, but it is a solid working example.

I did some tests of current sensing using ADC, although I am a little bit behind the schedule with that one. That is mainly caused by the implementation of PWM driver to iMXRT which was originally planned to take place during the last weeks of GSoC but was done during the previous week. But since iMXRT is the main platform target, I think it was right to make the implementation much earlier so the demos can be done with iMXRT based boards. This will make some free space during the next weeks where the current sensing can be finished.

michallenc, 2021-07-12:
Hello,

the fifth week of Google Summer of Code coding period was mainly about tickless mode support for iMXRT MCU. The source code [can be found here ](https://github.com/michallenc/incubator-nuttx/blob/imxrt-tickless/arch/arm/src/imxrt/imxrt_tickless.c). I still need to do some coding style changes before putting this to the mainline as well as some further testing. I run into some troubles with inaccuracies when I was using 32.7 kHz clock source as this frequency cannot be properly represented by CONFIG_USEC_PER_TICK. After discussion in NuttX mailing list, I switched to 16.6 MHz peripheral clock source, which divisions like 100 kHz or 10 kHz can be represented by CONFIG_USEC_PER_TICK. This change seems to fix the inaccuracies in the timer.

I also did some tests of ADC on iMXRT for current sensing and run into to problems when using multiple channels. There seems to be no DMA support available when using multiple channels, but I need to check my foundings with the chip´s reference manual and ADC driver source code. This might be crucial for the control of PMSM motors.

michallenc, 2021-07-19:
Hello,

my implementation of tickless mode for iMXRT MCU mentioned above was tested on real time DC motor control application and [merged into mainline during the past week](https://github.com/apache/incubator-nuttx/pull/4138). A problematic pattern in my code and in STM32 tickless mode code has been later noticed by @ppisa and [described in NuttX mailing list in detail](http://mail-archives.apache.org/mod_mbox/nuttx-dev/202107.mbox/browser). This problem does occur minimally as the overflow of the timer does not occur to often, but still should be prevented to happen. I plan to introduce a fix for both iMXRT and STM32, but firstly I need to do some tests of tickless mode with STM32.

After finishing the work on tickless mode, I started getting into the ADC support for iMXRT. As I mentioned last week, there are some problems when using multiple channels. It seems that sometimes the wrong channel is read and/or saved. The DMA support is missing in this case, but first option was to try avoid the problem by using multiple trigger control and read registers. That didn´t work out as only the first trigger register supports software trigger, the others support only hardware trigger. Althought this might not be a problem in the future as the ADC triggering should by done by PWM for the control of PMSM motor, it seems that using multiple channels with software trigger will need the DMA support.

michallenc, 2021-07-27:
Hello all,

during the last meeting, we discussed some changes to PWM and ADC drivers that would help the external application using them. All of them have already been merged to NuttX mainline. PR #4231 introduces ioctl command ANIOC_GET_NCHANNELS that returns the number of configured channels. This gives the applications the opportunity to be more flexible when it comes to multiple channel ADCs and they does not have to rely on user to specify the correct number of configured channels.

PR #4227 adds option to PWM drivers to prematurely leave the loops when -1 channel number occurs. This number can be used in applications at the end of channel list and indicate that all following channels are not used so the driver can skip them.

The changes that take the advantage of those updates are presented in pysimCoder PR [NuttX: update PWM and ADC blocks with recently added NuttX functionalities](https://github.com/robertobucher/pysimCoder/pull/21).

michallenc, 2021-08-02:
Hello,

this is going to be the last week of my Google Summer of Code 2021 coding period. During this week I plan to present documentation for running NuttX with pysimCoder, which would be placed in [Guides section](https://nuttx.apache.org/docs/latest/guides/index.html). I will also create representative configurations for some boards, probably Teensy 4.1 and Nucleo F446RE, so they can be easily used with pysimCoder.

During this week, I would also like to finish an examle with PMSM and do a similar video as was done with DC motor last month. I am currently finishing external ADC triggering with iMXRT MCU, the progress can be watched [in my forked repo](https://github.com/michallenc/incubator-nuttx/commits/teensy-dc). I´ve created [Clarke and Park transformations blocks for pysimCoder](https://github.com/michallenc/pysimCoder/commit/dc6a13f68dbbe2c3775c9e8095ccaf296e7234b6) which will be useful when it comes to PMSM control. There is still some optimalization tasks to be done before this gets merged into pysimCoder mainline, but it should be working.

My presentation about NuttX and pysimCoder has been approved to this year NuttX Workshop so I am looking forward to fully present to you the results of NuttX and pysimCoder colaboration in a few weeks.

closed 2025-05-09 by michallenc