apache/nuttx #3783: SPI SD card on Nucleo-F767zi unable to compile

ddkn, 2021-05-25:
Hello,

I am trying to get an SD card working on NuttX just over SPI. Currently, I am on commit 85b0fae, with the profile `nucleo-144:f767zi-nsh`. I was following the [Using SD card over SPI NuttX](https://www.youtube.com/watch?v=0CX-d9BFLrQ), however there are some changes to the menuconfig. I enabled as much as I could to simulate this step, i.e., setting SPI index to 1, enabling the SPI drivers, Device Drivers, disabling MMC, detect pins and write protect. However, when it tries to build I get issues with the `SDIO_MINOR` not being found,

[image]

Steps to reproduce:
```
./tools/configure.sh -l nucleo-144:f767-nsh
make menuconfig # change settings as seen in attached "config.txt"
make
```

[config.txt](https://github.com/apache/incubator-nuttx/files/6541807/config.txt)

Is there something obvious I am missing?

Cheers,

acassis, 2021-05-25:
Hi @ddkn currently SDIO_MINOR is only defined when native SD/MMC is enabled, see boards/arm/stm32f7/nucleo-144/src/nucleo-144.h

Double check you disabled CONFIG_STM32F7_SDMMC if you want to use SD card ove SPI

Seams like boards/arm/stm32f7/nucleo-144/src/stm32_sdio.c is compiled even when CONFIG_STM32F7_SDMMC is disabled, it is an error.

Also keep in mind that you need to prepare the SPI CS pin to work with your SDCard, please take a look at boards/arm/stm32/stm32f103-minimum/src/stm32_spi.c as reference, pay attention to CONFIG_MMCSD_SPI lines, etc. And then adapt boards/arm/stm32f7/nucleo-144/src/stm32_spi.c to have same support to enable CS pin.

ddkn, 2021-05-26:
Hi @acassis,

Are those your videos? Very helpful! So I did previously attach my .config as config.txt, and double checking it (and my build one just in case), it currently says:

`# CONFIG_STM32F7_SDMMC is not set`

I also double checked some relevant lines and explicitly set them to `=n`
```
CONFIG_MMCSD=y # stm32_sdio.c checks for this
...
CONFIG_STM32F7_HAVE_SDIO=n
...
CONFIG_STM32F7_SDMMC=n
CONFIG_STM32F7_SDMMC1=n # nucleo-144.h checks for this
CONFIG_STM32F7_SDMMC2=n # nucleo-144.h checks for this
```

I tried adding this to the top of stm32_sdio.c,
```
-#ifdef CONFIG_MMCSD
+#if defined(CONFIG_MMCSD) && defined(CONFIG_STM32F7_SDMMC)
```

But it errors out with `stm32_appinitialize.c:195 undefined reference to stm32_sdio_initialize`

I think before I am able to set the CS pin, maybe getting it to build first may be worth it.