apache/nuttx #3812: vfork() is broken.

patacongo, 2021-05-31:
The apps/testing/ostest vfork test no longer works in the protected mode.  A hardfault occurs in the test immediately following the barrier test.  The stm32f4discovery:kostest configuration was used for testing.  It has:

    # CONFIG_PRIORITY_INHERITANCE is not set
    CONFIG_ARCH_HAVE_VFORK=y
    CONFIG_SCHED_WAITPID=y

So this means that the failure is in the vfork() test:

    535 #if defined(CONFIG_PRIORITY_INHERITANCE) && !defined(CONFIG_DISABLE_PTHREAD)
    536       /* Verify priority inheritance */
    537
    538       printf("\nuser_main: priority inheritance test\n");
    539       priority_inheritance();
    540       check_test_memory_usage();
    541 #endif /* CONFIG_PRIORITY_INHERITANCE && !CONFIG_DISABLE_PTHREAD */
    542
    543 #if defined(CONFIG_ARCH_HAVE_VFORK) && defined(CONFIG_SCHED_WAITPID)
    544       printf("\nuser_main: vfork() test\n");
    545       vfork_test();
    546 #endif

Here is an analysis of the hardfault:

    up_assert: Assertion failed at file:armv7-m/arm_hardfault.c line: 135 task: ostest
    up_registerdump: R0: 00000002 20001f00 200021a0 200021a0 20001058 20001040 00000000 00000000
    up_registerdump: R8: 00000000 00000000 00000000 00000000 080067a3 20011288 080086e1 ffffffe8

Notice that the PC above is garbage.  The return address of 080086e1 is

    15611 0800863c <up_block_task>:
    ...
    15683  80086dc:       f000 fc6e       bl      8008fbc <arm_switchcontext>
    15684  80086e0:       bf00            nop

So it appears to have failed in attempting a context switch.  This would happen to the new context is corrupt.  My guess would be a bad set up of the vforked() context.  The commit that broke vfork() should be pretty easy to find.

    up_registerdump: xPSR: 01000000 PRIMASK: 00000001 CONTROL: 00000000
    up_registerdump: EXC_RETURN: ffffffe9
    up_dumpstate: sp:         200110f8
    up_dumpstate: stack base: 20010cb8
    up_dumpstate: stack size: 000007c8

Here is some context from the stack.  It does not reveal a lot because this is the logic executing on the vfork'ed thread so it has very little history.

    up_stackdump: 200110e0: 20011480 200110f8 200110e0 200110e0 200110f8 200111b4 20001040 00000000
    up_stackdump: 20011100: 20011108 08008531 000007c8 20010cb8 200110f8 20001e60 20011128 0800861b
    up_stackdump: 20011120: 00000087 20001fd4 00000087 08015240 0000000c 20001e60 20011140 08006815
    up_stackdump: 20011140: 00000087 08015240 20011150 0800142d 0000000c 00000000 200111b4 00000003

0800142d corresponds to the return from the call to _assert from arm_hardfault().

    up_stackdump: 20011160: 20011168 20010001 ffffffe6 200111b4 20011178 08002c55 200111b4 00000003

08002c55 is the call to arm_hardfault() in irq_dispatch().

    up_stackdump: 20011180: 20011188 00000003 00000000 080013b5 20011198 08000faf 200111b4 00000003

080013b5 is the address of arm_hardfault().
08000faf is return from the call to irq_dispatch() in arm_doirq().

    up_stackdump: 200111a0: 200111a8 00000000 00000000 08000f29 200111b8 20011288 00000001 20001058

08000f29 is the call to arm_doirq() from exception_common().

    up_stackdump: 200111c0: 20001040 00000000 00000000 00000000 00000000 00000000 00000000 ffffffe9
    up_stackdump: 200111e0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
    up_stackdump: 20011200: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
    up_stackdump: 20011220: 00000002 20001f00 200021a0 200021a0 080067a3 080086e1 ffffffe8 01000000
    up_stackdump: 20011240: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
    up_stackdump: 20011260: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
    up_stackdump: 20011280: 00000000 ffffffe9 00000000 00000000 00000000 00000000 00000000 00000000
    up_stackdump: 200112a0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
    up_stackdump: 200112c0: 00000000 00000000 00000001 20001f00 00000014 20001f00 08005a85 08008811

Most of the above is the state save from the exception.
08008811 is the return from the call to arm_fullcontextrestore() in up_exit()

    up_stackdump: 200112e0: 08008fba 01000000 00000000 00000000 00000000 00000000 00000000 00000000

08008fba is in arm_fullcontextrestore()

    up_stackdump: 20011300: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
    up_stackdump: 20011320: 00000000 00000000 00000000 20011330 20011338 00000000 00001340 20001e60
    up_stackdump: 20011340: 20011348 08003a9f 00000000 00000000 20011358 08003ac9 00000000 00000000

08003a9f is the address of exit()
08003ac9 is the return from _exit() in exit()

    up_stackdump: 20011360: 00000000 20002100 20011370 08005a97 00000000 00000001 20011398 080014cf

08005a97 is the return from exit in STUB_exit().
080014cf is the return to dispatch_syscall().

    up_stackdump: 20011380: 20001058 20001040 00000000 080270b1 00000018 00000000 00000000 0000000a
    up_stackdump: 200113a0: 200113a8 200113a8 200113bc 00000000 200113b8 0802623d 0802c11c 0000d300
    up_stackdump: 200113c0: 200113c8 080203d3 0001fea0 00000002 00000006 0000fff0 00001340 0001eb60
    up_stackdump: 200113e0: 20010c80 00000005 00000000 00000005 00000000 00000000 20011400 08020065
    up_stackdump: 20011400: 00000000 20010c80 00000005 0802018d 20011418 08008ef3 20010c80 00000005

08008ef3 is the return from sys_call4() in sys_call3().

    up_stackdump: 20011420: 0802018d 00000004 00000000 20011438 00000000 01000000 20010c80 00000005
    up_stackdump: 20011440: 0802018d 00000004 20011450 08008f3f 00000000 20010c80 00000005 0802018d

08008f3f is the return from sys_call3() in up_task_start().

    up_stackdump: 20011460: 20011468 08003a87 00000000 00000001 20001e60 00000005 00000000 00000000

08003a87 is the return from up_task_start() in nxtask_start().

masayuki2009, 2021-12-09:
@patacongo 
I think the regression happened on the master branch around June 2021.
Actually, both the latest master branch and the release/10.2 branch work fine.

patacongo, 2021-12-09:
> @patacongo I think the regression happened on the master branch around June 2021. Actually, both the latest master branch and the release/10.2 branch work fine.

OK.  then let's close this.

closed 2021-12-09 by patacongo