apache/nuttx #3826: arm-none-eabi-ld: Error: unable to disambiguate: -nostartfiles

AlexanderVasiljev, 2021-06-02:
I have faced the same issue as https://github.com/apache/incubator-nuttx/issues/3209.

I have just installed Ubuntu 21.04. The first build on new Ubuntu failed with this error.

LOG
```
LD: nuttx
arm-none-eabi-ld: Error: unable to disambiguate: -nostartfiles (did you mean --nostartfiles ?)
make[1]: *** [Makefile:156: nuttx] Error 1
```

GCC version
```
arm-none-eabi-gcc --version
arm-none-eabi-gcc (15:9-2019-q4-0ubuntu2) 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]
```

Ld version
```
rm-none-eabi-ld --version
GNU ld (2.36.1-0ubuntu1+14build1) 2.36.1
```

Ouss4, 2021-06-02:
Oh this issue is now on Debian based distros as well?

We've seen this (a general did you mean --option instead of -option) in arch based distros.  We couldn't see any references in the GNU project where -option is replaced with --option, so we assumed an issue in the Arch package.

Anyway, did you try to download the toolchain as described in https://github.com/apache/incubator-nuttx/issues/3209?

AlexanderVasiljev, 2021-06-02:
See my question on binutils
https://sourceware.org/pipermail/binutils/2021-June/116825.html

AlexanderVasiljev, 2021-06-02:
The flag nostartfiles is not a flag of ld. So it shouldn't appear in LDFLAGS += -nostartfiles

AlexanderVasiljev, 2021-06-03:
#3836

closed 2021-06-10 by davids5

hartmannathan, 2021-06-10:
This is caused by Binutils being updated to the 2.36.x line.

This is fixed by PR #3836

This is now documented at Release Notes for the (future) NuttX-10.2 release:

[https://cwiki.apache.org/confluence/display/NUTTX/NuttX+10.2#ld-now-called-through-gcc](https://cwiki.apache.org/confluence/display/NUTTX/NuttX+10.2#ld-now-called-through-gcc)