apache/nuttx #3839: Supporting BIOC_FLUSH in littlefs_sync_block()

amogh-samsung, 2021-06-03:
Hi, we are trying to port littlefs with your vfs layer code (lfs_vfs.c)

`littlefs_sync_block()` in `lfs_vfs.c` makes an MTD_IOCTL call with `cmd = BIOC_FLUSH`

For the case of imxrt1064 board, the ioctl calls are supported in function `imxrt_flexspi_nor_ioctl()`
This function does not seem to have BIOC_FLUSH support, hence it would return -ENOTTY

BIOC_FLUSH is supported in ftl.c with the help of rwbuffer.c
However, it's IOCTL base is _BIOC

Request you to kindly clarify the following please: -
1. With MTD_IOCTL, would the control flow reach ftl.c ?
2. Why is a BIOC command used with and MTD_IOCTL call ?
3. Is the BIOC_FLUSH supported properly or is it intended to return -ENOTTY ?

Kindly excuse and mistakes in my understanding.

xiaoxiang781216, 2021-06-03:
> Hi, we are trying to port littlefs with your vfs layer code (lfs_vfs.c)
>
> `littlefs_sync_block()` in `lfs_vfs.c` makes an MTD_IOCTL call with `cmd = BIOC_FLUSH`
>
> For the case of imxrt1064 board, the [...]

it's optional to support BIOC_FLUSH in MTD driver, the upper layer will convert ENOTTY to OK:
https://github.com/apache/incubator-nuttx/blob/master/fs/littlefs/lfs_vfs.c#L921-L937

> BIOC_FLUSH is supported in ftl.c with the help of rwbuffer.c
> However, it's IOCTL base is _BIOC
> 
> Request you to kindly clarify the following please: -
> 
> 1. With MTD_IOCTL, would the control flow reach ftl.c ?

No, the call chain is: ioctl->ftl_ioctl->MTD_IOCTL, not  ioctl->MTD_IOCTL->ftl_ioctl.

> 2. Why is a BIOC command used with and MTD_IOCTL call ?

Just to save one MTD_IOCTL number, but it's may better to add MTD_FLUSH, just like BIOC_XIPBASE and MTDIOC_XIPBASE.

> 3. Is the BIOC_FLUSH supported properly or is it intended to return -ENOTTY ?
> 

it's designed for the MTD layer which have a cache internally, and it's optional feature as mentioned before.

> Kindly excuse any mistakes in my understanding.

amogh-samsung, 2021-06-03:
Hello, thankyou for your prompt reply.
I understand that it is not necessary to support the lfs_cfg_s ->sync functionality if we are not buffering the write request data.

For Eg: - if CONFIG_FTL_WRITEBBUFFER is not set,
BIOC_FLUSH still returns -ENOTTY

However, you mentioned

> No, the call chain is: ioctl->ftl_ioctl->MTD_IOCTL, not ioctl->MTD_IOCTL->ftl_ioctl.

Could you help me understand this please?
Assuming an MTD driver registry for Littlefs, MTD_IOCTL(drvr->u.i_mtd, BIOC_FLUSH, ....) would call
(drvr->u.i_mtd)->ioctl(BIOC_FLUSH, ...) right?

For say, the imxrt1064 device, this ioctl is populated by vendor specific code in `boards/arm/imxrt/imxrt1064-evk/src/imxrt_flexspi_nor.c` which does not have `case: BIOC_FLUSH supported`

Again, pardon me for any mistakes in understanding

xiaoxiang781216, 2021-06-09:
@amogh-samsung to avoid the confusion, the new ioctl MTDIOC_FLUSH is added, please take a look: https://github.com/apache/incubator-nuttx/pull/3885

closed 2021-06-15 by gustavonihei