apache/nuttx #3898: posix_spawn handle the argv[0] in the wrong way

xiaoxiang781216, 2021-06-10:
From https://pubs.opengroup.org/onlinepubs/009695399/functions/posix_spawn.html:
```
The argument argv is an array of character pointers to null-terminated strings. The last member of this array shall be a null pointer and is not counted in argc. These strings constitute the argument list available to the new process image. The value in argv[0] should point to a filename that is associated with the process image being started by the posix_spawn() or posix_spawnp() function.
```
argv[0] should always point to the filename, but NuttX assume argv[0] point to the first argument and insert filename automatically:
https://github.com/apache/incubator-nuttx/blob/master/sched/task/task_setup.c#L572-L579

xiaoxiang781216, 2021-06-10:
posix_spawn(exec...) need remove this special insertion to confirm standard. But how to handle other similar but NuttX specific function(e.g. task_spawn, task_create...)? Changing to the same convention, or keep it as before but complicate the implementation and confuse people?

patacongo, 2021-06-11:
> 
> 
> posix_spawn(exec...) need remove this special insertion to confirm standard. But how to handle other similar but NuttX specific function(e.g. task_spawn, task_create...)? Changing to the same convention, or keep it as before but complicate the implementation and confuse people?

They cannot follow the same convention because there is no file name associated with task, only an entry point.  So it cannot work exactly the same way.  Something similar could be down like make argv[0] the task name, but that is kind of redundant.

xiaoxiang781216, 2021-06-12:
Do you think it's reasonable to enforce caller pass task name as the first argument so task_xxx API has the similar argv semantic as posix_spawn(exec...) API?

patacongo, 2021-06-12:
> Do you think it's reasonable to enforce caller pass task name as the first argument so task_xxx API has the similar argv semantic as posix_spawn(exec...) API? Or keep two set API has the different argv[0] interpretation?

I wouldn't change the task_xxx APIs at all.  I don't see any reason to do that and it would cause a lot of harm.

There are practical reasons to avoid doing this:

1. It is a big event, but you know that, and
2. It will break the code of EVERY use of NuttX.  I think that is wrong.  We should not cause NuttX user's such big problems unless there is a significant technical gain.  In there case, there is no technical gain.

And this does not help at all with the standardization of NuttX interfaces:

3. None of the task_xxx interfaces are standard.  There is no standard to comply with.
4. There is only one thing remotely like a standard.  task_create() (and some of the other task_xxx functions) derived originally from VxWorks interfaces.  They were more similar to the VxWorks interfaces in the past.  See https://docs.windriver.com/bundle/vxworks_7_application_core_os_sr0630-enus/page/CORE/taskLib.html
5. Putting a task name in argv[0] is not the same as putting a file name in argv[0].  There may be some superficial similarities, but it does not make them close to being the same.

So nothing is gained by making this awful change except for pissing off NuttX users.  The change would probably cause more harm than good.

xiaoxiang781216, 2021-06-14:
Ok, I keep task_xxx as before, but posix_spawn and execx follow the standard requirment, please review it here:
https://github.com/apache/incubator-nuttx/pull/3916
https://github.com/apache/incubator-nuttx-apps/pull/763

closed 2021-06-14 by patacongo