apache/nuttx #3915: Reopening a file does not clear I/O buffers.

fjpanag, 2021-06-12:
If I use I/O buffering (CONFIG_STDIO_DISABLE_BUFFERING = n), then the following piece of code fails.

```c
int c1;
int c2;

FILE * file = fopen("/file", "r");
c1 = getc(file);

file = freopen("/file", "rb");
c2 = getc(file);

assert(c1 == c2);
```

When the file is re-opened, the file index does not return to zero. `c2` gets the *second* character in the file, as it was found in the buffer (the file is not re-read).

Recompiling the code with `CONFIG_STDIO_DISABLE_BUFFERING` selected (no I/O buffering), the problem is solved.  
When I reopen the file, I get the first character of the file, as expected.

When a file is re-opened, shouldn't the I/O buffers be cleared?

pkarashchenko, 2021-12-17:
From the example I see that `file = freopen("/file", "rb");` is used, but `FAR FILE *freopen(FAR const char *path, FAR const char *mode, FAR FILE *stream);`, so the last parameter is missing in the example. Is the case incomplete?

fjpanag, 2022-11-04:
Fixed by #7345.

closed 2022-11-04 by fjpanag