apache/nuttx #3981: modularization or legotization of the project.

moe123, 2021-06-26:
Hello,

is it planed to split out a bit the code-base within nuggets ; for instance, such as, libc, libm, libcxx frontends making them standalone projects? indeed, those topics have already their own layer of complexity on top. Would be easier for people to intervene, indeed, the old school flat layout for OS has been an issue for decades i.e very difficult to onboard new people on specifics, collaborative layouts are the key.

xiaoxiang781216, 2021-06-26:
libm and libcxx come from other project, they are already in the standalone form. Since many libc component couple with NuttX implementation, it's hard to make it work without NuttX.

moe123, 2021-06-26:
hello, I think you don't understand.

xiaoxiang781216, 2021-06-26:
If so, please explain more about your suggestion.