apache/nuttx #4005: TCP_STOPPED is unused

yamt, 2021-07-01:
we have no code to set TCP_STOPPED.
it seems like a leftover of the original uIP api, namely uip_stop/uip_stopped, which have been removed in 2014.

yamt, 2021-07-01:
@patacongo have you kept this for reasons? otherwise i'd like to remove it.

yamt, 2021-07-01:
the commit which removed the api:
```
commit 5d7915e5efd92c779fe546ae1985fd977f82d2bb
Author: Gregory Nutt <gnutt@nuttx.org>
Date:   Mon Jun 30 19:03:01 2014 -0600

    NET: Remove uip_stopped and uip_stop
```

patacongo, 2021-07-01:
> @patacongo have you kept this for reasons? otherwise i'd like to remove it.

No, no reason.  Looks like it just wasn't completely removed.  The remaining logic does nothing since it tests the flag but there is no logic to set it.

I am not even sure what it was.  Perhaps it was the moral equivalent of shutdown()?

shutdown() is one of the important missing network APIs.

yamt, 2021-07-01:
i don't know what it was either.
it might be something similar to shutdown(SHUT_RD).
wrt implementing shutdown for us, my bet is that it's simpler to implement it from scratch than try to revive this uip_stop things.

patacongo, 2021-07-01:
> my bet is that it's simpler to implement it from scratch than try to revive this uip_stop things.

I agree.  I wasn't suggesting that.  The uip_stop thing was TCP specific.  shutdown() applies to all traffic  https://pubs.opengroup.org/onlinepubs/009695399/functions/shutdown.html

yamt, 2021-07-01:
> > my bet is that it's simpler to implement it from scratch than try to revive this uip_stop things.
> 
> I agree. I wasn't suggesting that. The uip_stop thing was TCP specific. shutdown() applies to all traffic https://pubs.opengroup.org/onlinepubs/009695399/functions/shutdown.html

well, i meant this TCP_STOPPED bit doesn't seem so useful for shutdown.
the uip_stop api is of course out of question.

fjpanag, 2022-11-04:
Since 5d7915e5efd92c779fe546ae1985fd977f82d2bb fixed this, shall we close the issue?

xiaoxiang781216, 2022-11-04:
@fjpanag here is the fix: https://github.com/apache/incubator-nuttx/pull/7528

fjpanag, 2022-11-04:
> @fjpanag here is the fix: #7528

Oh you **just** fixed it?  
Sorry, I thought it was already fixed, and that the issue was a left-over...

Thank you nevertheless...